padelt/temper-python #27: /etc/temper.conf scale and offset values not loading

Merric2015, 2015-03-02:
Hi,

First off, I really like the work you have done on this project & it has made using USB temperature devices so much easier.  I actually did some research before hand and specifically looked for a device I could use with your code.  Everything worked fine for me (following your instructions) apart from setting the scale and offset values so I thought I would highlight the issue I encountered & include my work around to help contribute back to the project.

On my Kubuntu 14.04LTS system I configured the /etc/temper.conf file as per the guide however my scale and offset values were not loaded or taken into account when calculating the temperature.  By adding some debug statements to the /temper-python/temperusb/temper.py file I was able to identify that the the self._ports value was actually an integer value being compared to string and the self._ports value needed to be cast as string for the comparison at line 112 to work.  I also needed to set the bus and port to be set exactly as per the output from temper-poll -p which was different to the explanation in the guide.

Here is my config and changes to work around the issue I encountered.

$python -V
Python 2.7.6

I'm using a temper device with the following id:
$lsusb
Bus 002 Device 004: ID 0c45:7401 Microdia

$lsusb -t
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 5000M
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 480M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=ehci-pci/2p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 2: Dev 3, If 0, Class=Mass Storage, Driver=usb-storage, 480M
        |__ Port 3: Dev 4, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
        |__ Port 3: Dev 4, If 1, Class=Human Interface Device, Driver=, 1.5M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=ehci-pci/2p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 1: Dev 3, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
        |__ Port 2: Dev 4, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M

$temper-poll -p
Found 1 devices
Device #0 (bus 2 - port 3): 20.5°C 68.9°F

/etc/temper.conf
2-3: scale = 1.0, offset = -4.6

With these settings (as stated above) I was able to read the temperature but the scale and offset values were not being recognised.  This was caused by the the self._ports value being passed as an integer when (on my system) it needed to be cast as string for the comparison.

/termperusb/temper.py
line 112 
                    if ports == self._ports:
changed to 
                    if ports == str(self._ports):
fixed the issue.

Here is the function the above line resides in.
    def set_calibration_data(self):
        """
        Set device calibration data based on settings in /etc/temper.conf.
        """
        self._scale = 1.0
        self._offset = 0.0
        try:
            f = open('/etc/temper.conf', 'r')
        except IOError:
            f = None
        if f:
            lines = f.read().split('\n')
            f.close()
            for line in lines:
                matches = re.match(CALIB_LINE_STR, line)
                if matches:
                    bus = int(matches.groups()[0])
                    ports = matches.groups()[1]
                    scale = float(matches.groups()[2])
                    offset = float(matches.groups()[3])
                    if ports == self._ports:
                        self._scale = scale
                        self._offset = offset

Changing line 112 resolved the issue for me & now the scale and offset values are correctly loaded and taken into account when printing the final temperature values.

Thanks again for the great work.

- Merric

surreydude, 2016-02-15:
I've just installed the latest versio and notice this workaround has been incorporated into the release code, however, my calibration data in /etc/temper.conf is still being ignored :(

user514, 2017-02-03:
same here. temper.conf seems ignored to me too. Same temps with and without /etc/temper.conf

jamesremuscat, 2023-11-18:
In case this helps anyone else stumbling across a similar issue - you need to specify both `scale` and `offset` in the config file as decimal numbers (e.g. `1.0` and not `1`) or the line won't be recognised.