padelt/temper-python #28: temperusb fails with newer pyusb

patricklauer, 2015-03-03:
pyusb-1.0.0_beta2 causes a failure on startup:

# python ./temperusb/cli.py 
Found 1 devices
Traceback (most recent call last):
  File "./temperusb/cli.py", line 81, in <module>
    main()
  File "./temperusb/cli.py", line 50, in main
    'temperature_c': dev.get_temperature(),
  File "/root/temper-python/temperusb/temper.py", line 152, in get_temperature
    self._interrupt_read()
  File "/root/temper-python/temperusb/temper.py", line 196, in _interrupt_read
    data = self._device.read(ENDPOINT, REQ_INT_LEN, interface=INTERFACE, timeout=TIMEOUT)
TypeError: read() got an unexpected keyword argument 'interface'

which seems to be fixed by removing the interface argument:

diff --git a/temperusb/temper.py b/temperusb/temper.py
index 06510a3..8d3e146 100644
--- a/temperusb/temper.py
+++ b/temperusb/temper.py
@@ -193,7 +193,7 @@ class TemperDevice(object):
         """
         Read data from device.
         """
-        data = self._device.read(ENDPOINT, REQ_INT_LEN, interface=INTERFACE, timeout=TIMEOUT)
+        data = self._device.read(ENDPOINT, REQ_INT_LEN, timeout=TIMEOUT)
         LOGGER.debug('Read data: {0}'.format(data))
         return data

amorphic, 2015-03-03:
There were other problems related to pyusb 1.0.0b2 as per #24. In the end we fixed the version to 1.0.0b1 because there didn't seem to be a workaround.

padelt, 2015-12-22:
So I just tried that again. Checked out master branch, essentially applied this patch:
```
diff --git a/setup.py b/setup.py
index ed992b7..24f7220 100644
--- a/setup.py
+++ b/setup.py
@@ -10,7 +10,7 @@ setup(
     long_description=open('README.md').read(),
     packages=['temperusb'],
     install_requires=[
-        'pyusb==1.0.0b1',
+        'pyusb>=1.0.0b2',
     ],
     entry_points={
         'console_scripts': [
diff --git a/temperusb/temper.py b/temperusb/temper.py
index 1e3a8ae..fef30f7 100644
--- a/temperusb/temper.py
+++ b/temperusb/temper.py
@@ -211,7 +211,7 @@ class TemperDevice(object):
         """
         Read data from device.
         """
-        data = self._device.read(ENDPOINT, REQ_INT_LEN, interface=INTERFACE, timeout=TIMEOUT)
+        data = self._device.read(ENDPOINT, REQ_INT_LEN, timeout=TIMEOUT)
         LOGGER.debug('Read data: {0}'.format(data))
         return data
```
Then (in a virtualenv) let `setup.py install` run and then
```
(for i in {1..30} ; do temper-poll 2>&1 | grep -i usberror ; done) | wc -l
8
```
That yielded 6, 8 and 8 USBErrors for 30 runs over 3 tries.
```
usb.core.USBError: [Errno None] error sending control message: Broken pipe
```
Using `pyusb 1.0.0b1` (as it is in master now) yielded 7,8 and 7 USBErrors. So after all, no real difference.

Fun fact: Using `pyusb 1.0.0b1` I get `libusb1`:
```
$ PYUSB_DEBUG=debug temper-poll 2>&1 | grep "using"
2015-12-22 14:23:37,832 INFO:usb.core:find(): using backend "usb.backend.libusb1"
```
Using `pyusb 1.0.0b2` I get `libusb0`:
```
$ PYUSB_DEBUG=debug temper-poll 2>&1 | grep "using"
2015-12-22 14:24:59,943 INFO:usb.core:find(): using backend "usb.backend.libusb0"
```

The cause seems to be an error while trying to find symbol `libusb_strerror`:
```
2015-12-22 14:26:19,325 ERROR:usb.backend.libusb1:Error loading libusb 1.0 backend
Traceback (most recent call last):
  File "build/bdist.linux-armv7l/egg/usb/backend/libusb1.py", line 880, in get_backend
    _setup_prototypes(_lib)
  File "build/bdist.linux-armv7l/egg/usb/backend/libusb1.py", line 453, in _setup_prototypes
    lib.libusb_strerror.argtypes = [c_uint]
  File "/usr/lib/python2.7/ctypes/__init__.py", line 378, in __getattr__
    func = self.__getitem__(name)
  File "/usr/lib/python2.7/ctypes/__init__.py", line 383, in __getitem__
    func = self._FuncPtr((name_or_ordinal, self))
AttributeError: /lib/arm-linux-gnueabihf/libusb-1.0.so.0: undefined symbol: libusb_strerror
2015-12-22 14:26:19,416 ERROR:usb.libloader:'OpenUSB library' could not be found
2015-12-22 14:26:19,418 ERROR:usb.backend.openusb:Error loading OpenUSB backend
2015-12-22 14:26:19,439 INFO:usb.core:find(): using backend "usb.backend.libusb0"
```
(This is on Raspbian on a Raspberry Pi, Linux 4.1.13-v7+ #826 SMP PREEMPT Fri Nov 13 20:19:03 GMT 2015 armv7l GNU/Linux)

That - in turn - stems from `pyusb 1.0.0b2` trying to use an API in `libusb1` that does not seem to be consistently available in all distributions of libusb: https://github.com/walac/pyusb/issues/57
Broken source in 1.0.0b2: https://github.com/walac/pyusb/blob/1.0.0b2/usb/backend/libusb1.py
The change that might have repaired it for us: https://github.com/walac/pyusb/commit/c46563c6b54acb9c2cbcbf7755f31fa3614e198b

I just tried it and the unreleased version 1.0.0rc1 of pyusb does in fact let my case run with `libusb1` as backend (same error rate). Have a look at (and test) https://github.com/padelt/temper-python/pull/40 .

I'll close this issue for now. I guess further discussion is best directed at the pull request.

closed 2015-12-22 by padelt