padelt/temper-python #29: Unchanged reading on temper-poll

surreydude, 2015-03-16:
Hi

Thanks for creating this library - I've installed it with one of my RDing Temper USB devices on a Raspberry Pi. All seems OK except that, on polling for the temperature every minute for the past 48 hours, I have noticed that my reading is a constant 16.5C

I have checked with another device and I have also used these devices on Win7 an they show variation. Are there any diagnostics I can turn on to establish more information about the sensor reading? Is it possible that I need to calibrate it?

Thanks

padelt, 2015-03-17:
Hi surreydude, I am not aware of any known issue that causes this - but in the distant past this was a behaviour I have seen myself. I could never reproduce it.
You could look for error messages. And obviously look for when the problem "goes away". Re-plug USB connection? Restart Pi? Or is it even "permanent"?

I find this very worrying because it makes the reading basically unreliable.

Sorry that I have no better news. If you can somehow reproduce it we might have a chance.

-- Philipp

amorphic, 2015-03-17:
Are you using the cli app or are you using the library as part of your own application?

If it's the latter you can turn on logging and you might better see what's going on. Something like:

```
LOGGER = logging.getLogger(__name__)     
LOGGER.setLevel(logging.DEBUG)
handler = logging.StreamHandler()
LOGGER.addHandler(handler)
```

surreydude, 2015-03-17:
padelt: I'm still getting 
<code>usb 1-1.2: usbfs: process 2295 (temper-poll) did not claim interface 0 before use</code>
messages in the log even though I have added a boot parameter for the <code>usbhid.quirks</code> as advised in the readme.

I've tried to compare two devices; dipped them in boiled water and can see the temperature reading rise (although I am unsure how accurate this is) and then fall to rest back at 16.5C.

I have also just ordered a more accurate temperature probe to try and establish the calibration; I am wondering if the scale is too low for what could be minor variations. Either way, the reading looks a bit "sticky".

surreydude, 2015-03-17:
Here's a snippet of the readings (you can see when the sensor was introduced to the water at 11:25):

<code>21,16.5,2015-03-17 11:21:01 </code>
<code>22,16.5,2015-03-17 11:22:01 </code>
<code>23,16.5,2015-03-17 11:23:01 </code>
<code>24,16.5,2015-03-17 11:24:01 </code>
<code>25,83.6,2015-03-17 11:25:01 </code>
<code>26,59.2,2015-03-17 11:26:01 </code>
<code>27,44.8,2015-03-17 11:27:01 </code>
<code>28,36.5,2015-03-17 11:28:01 </code>
<code>29,30.2,2015-03-17 11:29:01 </code>
<code>30,26.2,2015-03-17 11:30:01 </code>
<code>31,23.4,2015-03-17 11:31:01 </code>
<code>32,21.2,2015-03-17 11:32:01 </code>
<code>33,19.3,2015-03-17 11:33:32 </code>
<code>34,18.9,2015-03-17 11:34:01 </code>
<code>35,18.1,2015-03-17 11:35:01 </code>
<code>36,17.6,2015-03-17 11:36:01 </code>
<code>37,17.1,2015-03-17 11:37:01 </code>
<code>38,16.5,2015-03-17 11:38:01 </code>
<code>39,16.5,2015-03-17 11:39:01 </code>
<code>40,16.5,2015-03-17 11:40:01 </code>

surreydude, 2015-03-17:
amorphic: I'm using the library calling temper-poll from a ksh script. Should I put that code in <code>temper.py</code>?

padelt, 2015-12-21:
@surreydude The readouts on my sensors have noise on the order of 0.1°C and changes when touching it start to appear within 1-2 seconds. The error message you see (`did not claim interface 0 before use`) should be gone thanks to changes by @ps-jay - but those are not yet in my repo.

Could you please checkout the branch `ps-jay-reduce-kernel-syslogs` and try again? I listed a convenient way to test the branch in here: https://github.com/padelt/temper-python/pull/33#issuecomment-166363230

padelt, 2015-12-22:
I searched some more last night and came across http://www.thermohid.co.uk/ as well as some Amazon reviews (google for USB temper stickiness to find them). The effect of temperature readouts not changing for some time seems to be a hardware issue that other software has experienced as well.
Insofar we will not find a software solution.
Regarding the dmesg/syslog messages, please recheck with the (not yet released) version mentioned above.
I'll close this issue for now - feel free to reopen when you see something we can do against the "stickiness" please.

closed 2015-12-22 by padelt