padelt/temper-python #31: syslog full of usbfs logs

pziarsolo, 2015-06-03:
Hi,
I have been using temper-poll for a while. I have checked my logs and I realized that I have tree lines on it for each temper-poll call:

[4218339.022465] usb 5-1: usbfs: process 27266 (temper-poll) did not claim interface 0 before use
[4218339.026481] usb 5-1: usbfs: process 27266 (temper-poll) did not claim interface 1 before use
[4218339.324370] usb 5-1: reset low-speed USB device number 2 using uhci_hcd

It is possible to stop this logs? I have tried the usbhid.quirks=0x0c45:0x7401:0x4 in kernel options but it didn't work for me.

Thanks in advance
p.

ps-jay, 2015-06-15:
Are you using CentOS, or RHEL?

pziarsolo, 2015-06-15:
Nop, I am using debian 8

ps-jay, 2015-06-15:
I've experimented with a few lines, in particular doing some ```usb.util.claim_interface()``` calls and commenting out ```self._device.reset()```.

I'll clean it up and open a PR

jefonseca, 2015-11-09:
maybe this could help 
http://www.rsyslog.com/discarding-unwanted-messages/

I created a file in /etc/rsyslog.d/disabled.conf with this content:

:msg, contains, "input device has been disconnected" ~
:msg, contains, "TEMPerV1.4" ~
:msg, contains, "temper-poll" ~
:msg, contains, "reset low-speed USB device number 3 using ehci-pci" ~

pziarsolo, 2015-11-09:
@jefonseca . Thanks for the tip. It is very useful, now I can read syslog logs easily again :)

padelt, 2015-12-22:
Please retry with the current master branch. It should get rid of the messages. If not, feel free to reopen the issue!

closed 2015-12-22 by padelt