padelt/temper-python #32: Conversion from bytes steam to temp_c

ps-jay, 2015-06-15:
I wonder if we are over complicating the conversion from byte to str to temperature.

It looks like the first byte ```data[2]``` is the integer part of the degree, and the second byte ```data[3]``` is the fractional part of the degree - by dividing the value by 256.

In the [example spreadsheet](https://docs.google.com/spreadsheets/d/13pr1AK4CbSLBmcVBdw6WMunYdmL6VmFfINaOruiYeWM/edit?usp=sharing) you have the stream: ```80:04:1a:b0:19:70:2e:33```

Therefore:
```data[2]``` is ```0x1a``` which is 26 base 10 decimal
```data[3]``` is ```0xb0``` which is 176 base 10 decimal

And, ```data[3] / 256.0``` equals ```0.6875```

The "inner" column in the spreadsheet show the temp as 26,69... Hence I'd suggest:

```temp_c = data[2] + (data[3] / 256.0)```

However, I don't know how negative values of ```data[2]``` might appear in the stream.  Signed ints?  Twos complement?

The theory fits with a maximum high value of 125 deg.  (signed 8-bit ints max at 127)

padelt, 2015-12-22:
Yes there can be negative values and the current code expects it: `unpack('>h', data[2:4])` means we interpret two bytes as big-endian signed short. I need to check how the values look for temperatures <0°C. Your version is certainly equivalent for temperatures >=0°C. For negative temperatures, only one version can correct. Consider the data `0xFF 0xFB`:
```
>>> struct.unpack('>b','\xFF')[0] + (struct.unpack('>b','\xfb')[0] / 256.0)
-1.01953125
>>> struct.unpack('>h','\xFF\xFB')[0] / 256.0
-0.01953125
>>>
```
The first version is what you suggest, the second is what is currently implemented.

I will try to measure temperatures around 0°C to see the meaning of the values so we can decide what is "right".

padelt, 2015-12-25:
OK I did measure it with one internal and one external sensor type (different USB devices - each with one sensor). I cooled them down using Kontakt Chemie Kälte 75 spray. I'll show the 3rd and 4th byte along with the current temper-python output.
Device 1.2 (cooling down): 
```
0040 0.2°C
0000 0.0°C
FFE0 -0.1°C
FFC0 -0.2°C
FFA0 -0.4°C
FF20 -0.9°C
FF00 -1.0°C
FEEE -1.1°C
FEA0 -1.4°C
```
The readouts "made sense" as in "they did not jump around" but were somewhat continuous - keeping in mind the rather coarse resolution.

0xFFE0 interpreted as two's complement is -32, dividing that by 256 gives -0.125, which makes sense and is what we currently print. Using the second byte (0xE0) as the fractional part of a degree celcius would mean 224/256=0,875. What would work is using the first byte as signed char/byte (gives -1) and the second as unsigned char/byte, divide that by 256 and add them (as floats). Which (AFAICT) is a fancy and complicated way of interpreting the 2-byte word as a signed short - what we already do using `unpack('>h')/256.0`. So to sum up: I don't see the advantage of changing it.
Please correct me and reopen if I missed something!

closed 2015-12-25 by padelt