padelt/temper-python #37: Setup.py is not found

jcrick69, 2015-11-30:
Hi

I installed the package with
sudo apt-get install python-usb

If I try sudo python setup.py install

I get the error: can't open setup.py: [Errno 2] No such file or directory

I have confirmed the package is installed.

Note I am using a Raspberry Pi 2

Any help greatly appreciated.

Thanks, Jerry

amorphic, 2015-12-02:
Jerry,

To install the package from source you need to first clone it from github like this:
```
git clone git@github.com:padelt/temper-python.git
cd temper-python
sudo python setup.py install
```
An even easier way is to use pip to download and install the package from [PyPI](https://pypi.python.org/) (the Python Package Index):
```
sudo apt-get install pip
pip install temperusb
```
Cheers,
James

jcrick69, 2015-12-06:
Hi James

Thanks for he help but,

neither of these methods work.

The first,in the clone step, gives an error saying the remote hung up
unexpectedly.

The second, in the apt-get, says unable to locate package pip

Any ideas gratefully accepted...

Thanks, Jerry

-- 
Jerry Crick

<http://uk.linkedin.com/pub/jerry-crick/16/194/428>

Skype: jerrycrick69

T: 01582 612 604
F: 01582 612 605
M: 07771 573 762
E: jcrick@bvuz.biz

Website: www.bonessvets.co.uk
Facebook: www.facebook.com/bonessvets
Twitter: www.twitter.com/bonessvet

125 Bedford Road, Barton le Clay, Bedfordshire MK45 4LP

padelt, 2015-12-06:
Hi Jerry,
git clone likely fails because you do not have SSH keys set up with Github. You don't need to though.
Look at the github page and copy the https-URL for cloning the repo.
git@... will be replaced by https://...
Can't look it up exactly right now.

> sudo apt-get install pip
That should most likely be
> sudo apt-get install python-pip

Hope it gets you going...

Philipp

Vom padelt gesendet

padelt, 2015-12-21:
Jerry, I also just noticed that you tried
```
sudo apt-get install python-usb
```
(note the `python-usb`-package)

This repository is about a package called `temperusb` (on PyPI, which is the Python Packag Index that `setuptools` and `pip` use). It has little to do with `python-usb`.

I'll close this issue - feel free to comment and reopen.

Philipp

closed 2015-12-21 by padelt