padelt/temper-python #41: Upload PyPI package version 1.3.1

padelt, 2016-01-05:
@amorphic Could you please upload the new version 1.3.1 to PyPI?

amorphic, 2016-01-07:
Hi Philipp!

I built and uploaded `1.3.1` as requested but when I install it to a new 2.7.6 virtualenv the `pyusb>=10.0.rc1` dependency causes problems:

```
Could not find a version that satisfies the requirement pyusb==1.0.0rc1 (from versions: 1.0.0a2, 1.0.0a2, 1.0.0a3, 1.0.0a3, 1.0.0b1, 1.0.0b2)
```

Have you tried that latest code with one of these versions of pyusb? Perhaps we need to release a `1.3.2` with the `pyusb` pinned to whichever of these alpha/beta versions is known to be working?

padelt, 2016-01-07:
That's the problem: None of these are working. rc1 is the first one that does.
When I `python setup.py install` it like this, it works:
```
pa@philipp:~/temper$ git clone https://github.com/padelt/temper-python.git
pa@philipp:~/temper$ virtualenv v1
pa@philipp:~/temper$ . v1/bin/activate
(v1)pa@philipp:~/temper/temper-python$ python setup.py install
(v1)pa@philipp:~/temper/temper-python$ temper-poll
Found 0 devices
```
When using `pip install temperusb`, I also get the problem:
```
(v2)pa@philipp:~/temper$ pip install temperusb
Downloading/unpacking temperusb
  Downloading temperusb-1.3.1.tar.gz
  Running setup.py (path:/home/pa/temper/v2/build/temperusb/setup.py) egg_info for package temperusb

Downloading/unpacking pyusb>=1.0.0rc1 (from temperusb)
  Could not find a version that satisfies the requirement pyusb>=1.0.0rc1 (from temperusb) (from versions: 1.0.0a2, 1.0.0a2, 1.0.0a3, 1.0.0a3, 1.0.0b1, 1.0.0b2)
Cleaning up...
No distributions matching the version for pyusb>=1.0.0rc1 (from temperusb)
Storing debug log for failure in /home/pa/.pip/pip.log

```
This is with `pip` version 1.5.4 and `pyusb` can "only" be fulfulled using the git+https-Link in the `setup.py`. It is not yet on PyPI. Is that the problem maybe!?
How can we force pip to honor the `dependency_links` setting when installing from PyPI?

ps-jay, 2016-01-07:
We probably should write to the pyusb maintainers, and ask them to make a release on PyPi...

padelt, 2016-01-07:
Yes I think that is a good idea. I'll open an issue over at https://github.com/walac/pyusb/issues asking for an rc1-release.

padelt, 2016-01-07:
So the answer is in: He tries to release it ASAP, but I think we should not hold our breath. Maybe someone can come up with another way to get rc1 working when temperusb is installed from PyPI. Any ideas?

ps-jay, 2016-04-15:
pyusb 1.0.0rc1 was released Feb 16, and works as expected.

We can go ahead and release temperusb 1.4.1 now (and 1.3.1 for completeness I suppose!)

Python 2:
```
(py2) root@raspberrypi:~/temper-python# pip install pyusb
Collecting pyusb
  Downloading pyusb-1.0.0rc1.tar.gz (53kB)
    100% |████████████████████████████████| 61kB 495kB/s
Building wheels for collected packages: pyusb
  Running setup.py bdist_wheel for pyusb ... done
  Stored in directory: /root/.cache/pip/wheels/e5/52/fd/eedc7ae38c634040e5bc7e6f9188a2efcc8500acaac28248f7
Successfully built pyusb
Installing collected packages: pyusb
Successfully installed pyusb-1.0.0rc1
(py2) root@raspberrypi:~/temper-python# python -m temperusb.cli
Found 1 devices
Device #0: 23.1°C 73.6°F
```

Python 3:
```
(py3) root@raspberrypi:~/temper-python# pip install pyusb
Collecting pyusb
  Using cached pyusb-1.0.0rc1.tar.gz
Building wheels for collected packages: pyusb
  Running setup.py bdist_wheel for pyusb ... done
  Stored in directory: /root/.cache/pip/wheels/e5/52/fd/eedc7ae38c634040e5bc7e6f9188a2efcc8500acaac28248f7
Successfully built pyusb
Installing collected packages: pyusb
Successfully installed pyusb-1.0.0rc1
(py3) root@raspberrypi:~/temper-python# python -m temperusb.cli
Found 1 devices
Device #0: 23.1°C 73.6°F
```

amorphic, 2016-04-15:
Hey guys. Glad to hear we finally have a `pyusb` that we can use!

I'm in the middle of a couple of things right now but I'll do my best to get the releases out over the weekend.

ps-jay, 2016-04-15:
We should probably get the tags back in the repo (#51) before doing so...

padelt, 2016-06-12:
So, the old tags won't come back. The new releases will both be tagged and the commit id noted in `CHANGELOG.md`.
@amorphic Could you update PyPI with the new version 1.5.1? `setup.py` should now pull `pyusb` from PyPI instead of using the git checkout (that failed when `pip` installing `temper-python`).

amorphic, 2016-06-16:
Hi @padelt! Sorry I've taken a while to get back to you. Really busy week.

Just about to go to bed but I'll get `1.5.1` pushed to PyPI tomorrow.

amorphic, 2016-06-21:
Sorry for the delay all. `1.5.1` is [now on PyPI](https://pypi.python.org/pypi/temperusb/1.5.1).

padelt, 2016-06-22:
Great, thanks James!

closed 2016-09-07 by padelt