padelt/temper-python #45: Python 3.x support?

willfurnass, 2016-01-17:
A few things that need to be handled differently in Python 3.x:

 - `temper.py` features `long` hex literals like `0x0c45L`.  Python 3 does not have `long`s (not needed as `int`s are arbitrary precision).
 - do `COMMANDS` need to be byte strings?
 - the part of `TemperDevice.get_temperature` that interprets device responses may need to change
 - `range`s are generator expressions so in some cases need to be explicitly converted to `list`s
 - possibly other things

willfurnass, 2016-01-19:
See PR #46

padelt, 2016-01-20:
Thanks @willfurnass for the work. I added some metadata to `setup.py` and bumped the PyPI version to indicate the new Python 3 support. Remember to please send a PR for the contributor line in `README.md` if you want.

closed 2016-01-20 by padelt