padelt/temper-python #48: Raspbian compatible?

surreydude, 2016-02-15:
Hi
I'm using this library for reading USB TEMPer devices plugged into Raspberry Pi (vB1 & 2). Has it been tested on Raspbian? It works but I get some errors using `temper-poll -p`

padelt, 2016-02-15:
Works perfectly. In fact, it is the system I am using them on. Now on a 2B, before on a stock Raspi 1.
If you still get the errors, please reopen and include the actual output.
It is also interesting to know how you installed temper-python to know the version.

closed 2016-02-15 by padelt

surreydude, 2016-02-15:
I downloaded v1.4.0

On my Piv2 running `temper-poll -p` yields:

`Found 1 devices
Traceback (most recent call last):
  File "/usr/local/bin/temper-poll", line 9, in <module>
    load_entry_point('temperusb==1.4.0', 'console_scripts', 'temper-poll')()
  File "build/bdist.linux-armv7l/egg/temperusb/cli.py", line 66, in main
KeyError: 'bus'
`

padelt, 2016-02-15:
I can confirm this. The cause was an oversight of the last refactoring round. I just fixed the issue. Please try
```
git pull
python setup.py install
```
from your existing `temper-python`-directory. `git` should pull one new change.
Please reopen this issue if it does not fix your problem!

surreydude, 2016-02-15:
Fixed - thanks!