padelt/temper-python #49: temper-poll works once, then times out on subsequent calls

Braklet, 2016-02-15:
The Python poll application works, but only one time.  On subsequent calls, the device is detected but I get a handler error and the poll times out.

```
rlr@Cube:~$ temper-poll
Found 1 devices
Device #0: 25.4Â°C 77.7Â°F
rlr@Cube:~$ temper-poll
Found 1 devices
No handlers could be found for logger "temperusb.temper"
Traceback (most recent call last):
  File "/usr/local/bin/temper-poll", line 9, in <module>
    load_entry_point('temperusb==1.4.0', 'console_scripts', 'temper-poll')()
  File "build/bdist.linux-x86_64/egg/temperusb/cli.py", line 48, in main
  File "build/bdist.linux-x86_64/egg/temperusb/temper.py", line 254, in get_temperatures
  File "build/bdist.linux-x86_64/egg/temperusb/temper.py", line 194, in get_data
  File "build/bdist.linux-x86_64/egg/temperusb/temper.py", line 293, in _interrupt_read
  File "build/bdist.linux-x86_64/egg/usb/core.py", line 988, in read
  File "build/bdist.linux-x86_64/egg/usb/backend/libusb1.py", line 851, in intr_read
  File "build/bdist.linux-x86_64/egg/usb/backend/libusb1.py", line 936, in __read
  File "build/bdist.linux-x86_64/egg/usb/backend/libusb1.py", line 595, in _check
usb.core.USBError: [Errno 110] Operation timed out
rlr@Cube:~$
```

I can repeat this forever by removing then reinserting my USB sensor.

Not sure where to start breaking this down... can you help?
(Installed in Ubuntu 14.04 x86-64 system, kernel 3.13.0-77-generic.

Thanks!

padelt, 2016-02-15:
Not sure what causes this and not sure if we can do anything against it. My gut feeling is that your temper device is slightly different from mine. Could you show us the output of `dmesg`, `lsusb`, `python -c "import usb;print usb.__version__"` and the part of `lsusb -v` that describes the temper device? Maybe that shines some light on the root cause.

There is also a pull request (still open) that adds support for newer(?) versions of the TEMPer device: https://github.com/padelt/temper-python/pull/43
@roques problem sounds similar to yours. You could try that version:
```
git clone -b TEMPerV1.4 https://github.com/roques/temper-python/ temper-roques-TEMPerV1.4
cd temper-roques-TEMPerV1.4
virtualenv venv
. venv/bin/activate
python setup.py install
which temper-poll
PYUSB_DEBUG=debug temper-poll
```
The `PYUSB_DEBUG` environment variable will make `pyusb` emit lots of info including something like this, which may be interesting to debug:
```
INFO:usb.core:find(): using backend "usb.backend.libusb1"
```

Braklet, 2016-02-15:
Thanks for the reply.  I expected to have to play with this thing...

DMESG:
```
[ 3375.613107] usb 3-3: USB disconnect, device number 11
[ 4678.514909] usb 3-3: new low-speed USB device number 12 using xhci_hcd
[ 4678.537063] usb 3-3: New USB device found, idVendor=0c45, idProduct=7401
[ 4678.537072] usb 3-3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 4678.537077] usb 3-3: Product: TEMPerV1.4
[ 4678.537081] usb 3-3: Manufacturer: RDing
[ 4678.537279] usb 3-3: ep 0x81 - rounding interval to 64 microframes, ep desc says 80 microframes
[ 4678.537291] usb 3-3: ep 0x82 - rounding interval to 64 microframes, ep desc says 80 microframes
[ 4678.540654] input: RDing TEMPerV1.4 as /devices/pci0000:00/0000:00:14.0/usb3/3-3/3-3:1.0/input/input23
[ 4678.540972] hid-generic 0003:0C45:7401.000C: input,hidraw0: USB HID v1.10 Keyboard [RDing TEMPerV1.4] on usb-0000:00:14.0-3/input0
[ 4678.543415] hid-generic 0003:0C45:7401.000D: hiddev0,hidraw1: USB HID v1.10 Device [RDing TEMPerV1.4] on usb-0000:00:14.0-3/input1
```
LSUSB:
```
Bus 002 Device 002: ID 8087:8001 Intel Corp.
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 8087:8009 Intel Corp.
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 003: ID 1997:2433
Bus 003 Device 012: ID 0c45:7401 Microdia
Bus 003 Device 004: ID 09ae:2007 Tripp Lite
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
LSUSB -v:
```
rlr@Cube:~$ lsusb -vd 0c45:7401

Bus 003 Device 012: ID 0c45:7401 Microdia
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0         8
  idVendor           0x0c45 Microdia
  idProduct          0x7401
  bcdDevice            0.01
  iManufacturer           1 RDing
  iProduct                2 TEMPerV1.4
  iSerial                 0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           59
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.10
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      65
         Report Descriptors:
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval              10
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.10
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      41
         Report Descriptors:
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval              10
Device Status:     0x0000
  (Bus Powered)
```
Python library version:
```
rlr@Cube:~$ python -c "import usb;print usb.__version__"
1.0.0rc1
```
I'll take a look at the forked library, too.  Thanks for the tips!

padelt, 2016-02-15:
Chances are you are seeing the exact same problem and that @rogues changes will help. IF they do: Please tell me where you bought the device. I would like to have one too so I can properly test and incorporate @rogues changes!

Braklet, 2016-02-15:
Well I'll be... looks like @roques problem was indeed same as mine:

```
(venv)rlr@Cube:~/Devices/TemperUSB/Source/temper-roques-TEMPerV1.4$ temper-poll
Found 1 devices
Device #0: 23.2Â°C 73.7Â°F
(venv)rlr@Cube:~/Devices/TemperUSB/Source/temper-roques-TEMPerV1.4$ temper-poll
Found 1 devices
Device #0: 23.2Â°C 73.7Â°F
(venv)rlr@Cube:~/Devices/TemperUSB/Source/temper-roques-TEMPerV1.4$ temper-poll
Found 1 devices
Device #0: 23.2Â°C 73.7Â°F
(venv)rlr@Cube:~/Devices/TemperUSB/Source/temper-roques-TEMPerV1.4$ temper-poll
Found 1 devices
Device #0: 23.2Â°C 73.7Â°F
(venv)rlr@Cube:~/Devices/TemperUSB/Source/temper-roques-TEMPerV1.4$ temper-poll
Found 1 devices
Device #0: 23.2Â°C 73.7Â°F
(venv)rlr@Cube:~/Devices/TemperUSB/Source/temper-roques-TEMPerV1.4$ temper-poll
Found 1 devices
Device #0: 23.2Â°C 73.7Â°F
(venv)rlr@Cube:~/Devices/TemperUSB/Source/temper-roques-TEMPerV1.4$
```
Got some weird-ish behavior when I went back and installed @roques version on top of the original:
```
zipimport.ZipImportError: bad local file header in /usr/local/lib/python2.7/dist-packages/temperusb-1.4.0-py2.7.egg
rlr@Cube:~/Devices/TemperUSB/Source/temper-roques-TEMPerV1.4$ temper-poll
Found 1 devices
WARNING:temperusb.temper:Encountered [Errno 110] Operation timed out, resetting <DEVICE ID 0c45:7401 on Bus 003 Address 012> and trying again.
Device #0: 23.2Â°C 73.7Â°F
rlr@Cube:~/Devices/TemperUSB/Source/temper-roques-TEMPerV1.4$ temper-poll
Found 1 devices
Device #0: 23.2Â°C 73.7Â°F
rlr@Cube:~/Devices/TemperUSB/Source/temper-roques-TEMPerV1.4$ temper-poll
Found 1 devices
Device #0: 23.2Â°C 73.7Â°F
rlr@Cube:~/Devices/TemperUSB/Source/temper-roques-TEMPerV1.4$ temper-poll
Found 1 devices
Device #0: 23.2Â°C 73.7Â°F
```
But it seems to be working after multiple calls.

Am I unique in getting the 'Â' character in my output?  Haven't seen that in anybody else's after some minor Googling.

Thanks again for your driver and all your help.

Braklet, 2016-02-15:
Ah, just caught your second comment.  I purchased this one at Amazon:

http://www.amazon.com/gp/product/B009YRP906

I'm guessing my v1.4 is a key?

surreydude, 2016-02-15:
The 'Â' is due to your putty/ssh encoding; try setting it to UTF-8

Braklet, 2016-02-15:
@surreydude, that was it exactly.  Changed remote character set from ISO-8859-1 to UTF-8 eliminated the additional output.  Thanks!

surreydude, 2016-02-15:
You're welcome. There's something I know the answer to - yay!

padelt, 2016-02-15:
Thanks for the pointer to Amazon. Unfortunately, it is not delivered to Germany... :-/
But I did find one on amazon.de that has a mentioning of 1.4 in the comments, so I just ordered that.
Hopefully, the one delivered will actually show the same behaviour so we can finally merge these changes.

`temper-usb` prints UTF-8 and your terminal seems to interpret that as something different. You can see the current locale using `locale | grep LANG`. Mine says `LANG=en_GB.UTF-8`. On Ubuntu, you should be able to use `sudo dpkg-reconfigure locales` to select a UTF-8 locale.

Braklet, 2016-02-15:
Thanks again for the help.  I am interested in buying more of these things for various projects, so will keep an eye on your progress.

closed 2016-02-15 by Braklet

roques, 2016-02-15:
Just to confirm what you already discovered: This looks *exactly* like the problems I encountered. After one successful poll with the unpatched ``temper-poll`` the TEMPer gets stuck and future polls time out until the TEMPer is reset (or un- and re-plugged). I guess that the timeout after installing my branch into ``/usr/local`` was caused by you executing the unpatched ``temper-poll`` once and thus "hanging" the TEMPer. My branch will automatically reset hanging TEMPers, Once reset, polling with the patched code should work fine, without requiring further resets.

Braklet, 2016-02-15:
In any case, thanks for the patch.  Worked great on my v1.4 TEMPer.
I need to play around with this infrastructure a little more, but I'll probably buy a few more TEMPers once I get this working as I'd like.
Braklet

padelt, 2016-02-17:
Hmm, just got the sensor bought from Amazon under ASIN B009RETJIO from seller "zjchao". "Unfortunately" it works fine and it is not riddled with the v1.4 problem. It identifies as TEMPer1F_V1.3. Sending it back right now and ordering another one...

pmackinney, 2016-03-23:
For the device database: My device is labeled TEMPer1 -40oC ~ +120oC
From lsusb: Bus 003 Device 011: ID 0c45:7401 Microdia TEMPer Temperature Sensor

On Ubuntu 14.04, I was running temperpoll 1.2.3 without error.

Upgraded to Ubuntu 16.04, installed temperpoll 1.4.1 from Git Hub, had to download and install pyusb-1.0.0rc1 to satisfy dependencies. Immediately encountered the run once issue. Roques' patch works great.

josvos, 2016-03-27:
@padelt: I ordered my device (TEMPerV1.4) from [DX.com](http://www.dx.com/), for around 7 euro incl. shipping I got it in the Netherlands (shipping may take a few weeks though...), so you can get it in Germany too I guess: [DX product page](http://www.dx.com/p/temper-usb-thermometer-temperature-recorder-for-pc-laptop-81105)

I had the same errors and I tries @roques version and that now works for me.