padelt/temper-python #50: git pull failing for pyusb dependency

surreydude, 2016-02-22:
Hi

Following the my issue last week and your update, I have updated one of my other Pi installations using:

`git pull`

then

`python setup.py install`

and I'm getting a problem with the pyusb dependency:

`
Installing temper-poll script to /usr/local/bin
Installing temper-snmp script to /usr/local/bin

Installed /usr/local/lib/python2.7/dist-packages/temperusb-1.4.1-py2.7.egg
Processing dependencies for temperusb==1.4.1
Searching for pyusb>=1.0.0rc1
Best match: pyusb 1.0.0rc1
Downloading git+https://github.com/walac/pyusb.git#egg=pyusb-1.0.0rc1
error: Download error for git+https://github.com/walac/pyusb.git: unknown url type: git+https
`

ps-jay, 2016-02-22:
Pretty sure that's the work around for using the pre-release pyusb.

What version of git do you have installed?
What version of pip?

- Phil

padelt, 2016-02-22:
Some googling indicates that a newer version of setuptools is needed.
Would versions are you running? Here are mine (that work):
```
pa@plug2:~/bin$ easy_install --version
setuptools 18.5 from /usr/local/lib/python2.7/dist-packages (Python 2.7)
pa@plug2:~/bin$ pip -V
pip 7.1.2 from /usr/local/lib/python2.7/dist-packages (python 2.7)
```
Somthing like 
```
sudo pip install setuptools –upgrade
```
may be needed.

-- Philipp

surreydude, 2016-02-22:
```easy_install --version
distribute 0.6.24dev-r0
pip -V
pip 8.0.2 from /usr/local/lib/python2.7/dist-packages/pip-8.0.2-py2.7.egg (python 2.7)
sudo pip install setuptools -upgrade

Usage:
  pip install [options] <requirement specifier> [package-index-options] ...
  pip install [options] -r <requirements file> [package-index-options] ...
  pip install [options] [-e] <vcs project url> ...
  pip install [options] [-e] <local project path> ...
  pip install [options] <archive url/path> ...

no such option: -u
```

ps-jay, 2016-02-22:
> pip install [options] [package-index-options] ...
> pip install [options] -r [package-index-options] ...

To upgrade: pip install -U <package>

surreydude, 2016-02-22:
@ps-jay 

Thanks, that helped, I now get:

`setuptools 20.1.1 from /usr/local/lib/python2.7/dist-packages (Python 2.7)`

but now `sudo python setup.py install` is returning:

Traceback (most recent call last):
  File "setup.py", line 1, in <module>
    from setuptools import setup
ImportError: No module named setuptools

surreydude, 2016-02-22:
I've obviously got a roasted python installation - I've just nuked it and am reinstalling Raspbian jesse with python 2.7.9

closed 2016-02-22 by surreydude