padelt/temper-python #51: Tags not on github

ps-jay, 2016-04-13:
Hi Philipp,

the tags seem to have disappeared from the Github repo...

padelt, 2016-04-15:
I have no idea where they went!? Any ideas? TBH, I don't even remember what was in there.

ps-jay, 2016-04-15:
If you've got a local copy of the repo with tags, you might be able to recreate them by inspecting the commit ID of each...  (or easier, git push --tags)

If you don't have a local copy of the tags, then you'll just have to infer which commits were tagged (probably ones that incremented the version in `setup.py`?)

padelt, 2016-04-15:
I'll have a look at home but it seems I don't have any checked out version including tags.
If nobody has a list of commits associated with the tags, maybe we'll just start tagging them now.
Sometimes I miss Mercurial, where shooting your foot was/is considerably harder.

amorphic, 2016-04-19:
Hey guys,

So I just found some free time and went to package up a new release and saw this.

Are we safe to just tag the existing codebase `1.4` and bundle a release? Would you prefer I jump to `1.5` and we retrofit `1.4` later if we can divine the tags?

FWIW in my projects I've taken to keeping a `CHANGELOG` as per [these guidelines](http://keepachangelog.com/). It makes it very easy to see at a glance what has been added with each version (and which changes are outstanding waiting to be released in a new version).

@padelt if you like I could add a `CHANGELOG` to the project or I could leave it to you. Just let me know.

padelt, 2016-04-20:
Hi James,
AFAICT, I have no source for the missing tags. :-/
I'd say we call the current version 1.5.0 (which is already in the
setup.py file) and I'll tag the current commit
c5a82f1359e8bf82ea31c655f610d28091f070fb v1.5.0 later today.
I'll also try to at least tag 1.3.1 as it is today on PyPI by comaring
the files.

@amorphic It would be really nice if you could start the changelog!

-- Philipp

ps-jay, 2016-04-20:
After tagging, we could always update CHANGELOG.md to record the SHA1 where the tag was made.

padelt, 2016-04-20:
I thought about that and was unsure where that information should go. Is there a definitive answer? Or at least opinions?
Feel free to reopen if needed.

closed 2016-04-20 by padelt