padelt/temper-python #52: Import not working

ps-jay, 2016-04-13:
Hi all,

since this change: https://github.com/padelt/temper-python/commit/59912734bea136a766d1d30e3c080ae220598b16, altered: ```from temperusb.temper import TemperHandler```, to: ```from .temper import TemperHandler```

I cannot run ```cli.py``` directly:

```
 $ PYTHONPATH=. python temperusb/cli.py -S 2 -s all -p
Traceback (most recent call last):
  File "temperusb/cli.py", line 5, in <module>
    from .temper import TemperHandler
ValueError: Attempted relative import in non-package
```

Reverting that import to the original seems to have no functional impact in the ```python setup.py install && temper-poll``` type scenario.

@willfurnass what's the driver behind the change?  If it's reverted will it break something for you?

willfurnass, 2016-04-14:
Python 3 support.  If the `.` suffix is removed from `temper` then running `temper-poll` after installing `temper-python` using Python 3 + pip will cause problems. `cli.py` (exposed as `temper-poll` when installed with pip) will look for `TemperHandler` under the absolute path `temper` yet there is no installed package called `temper`; `cli.py` instead needs to look for `TemperHandler` under the relative path `.temper` (a module in the same directory as `cli.py`).

Is there a strong interest in being able to run `cli.py` directly without installing `temper-python` as a package?

ps-jay, 2016-04-15:
I'm not sure about strong interest - but I like it ; )

ps-jay, 2016-04-15:
Okay - I worked it out...  I need to run it as a module like so: ```python -m temperusb.cli```

```
(py2) root@raspberrypi:~/temper-python# python -m temperusb.cli
Found 1 devices
Device #0: 23.1°C 73.6°F
```

closed 2016-04-15 by ps-jay