padelt/temper-python #59: Temperature reading near zero read near 255.

marklewis3000, 2016-05-04:
I put my TEMPer device in an ice bath. I set the /etc/temper.conf file scale=1.0 and offset=0.0 to uncalibrate it. As it cools in the ice bath, the degrees C readings go as 1.3, 0.8, 0.3, 255.9, 255.6. In other words, when it reaches 0.0 degrees, you have to subtract off 256 (still uncalibrated). These values of course mess with the degrees F as well, which respectively show as 34.4, 33.4, 32.6, 492.7, 492.0 !!

Since the advertise device range is -40 to 120 degrees C, I think you need to have a check in the code, after fetching the data at line 259 of temper.py,

-         data = self.get_data()

something like

-         if data >= 128:
                 data = data - 256

marklewis3000, 2016-05-09:
Details of the TEMPer device are given here:
https://sigrok.org/wiki/RDing_TEMPer_Gold#Resources
Which says, theTemperature sensor is a NXP LM75A.
Data sheet can be found here:

smeek, 2017-01-09:
I've got a pull request (https://github.com/padelt/temper-python/pull/68) open for a fix to this issue.

padelt, 2017-03-31:
This should be fixed in the new master-version. This is not released on PyPI yet.
@marklewis3000 Please test again and re-open the issue if there are further problems!

closed 2017-03-31 by padelt