padelt/temper-python #60: trouble installing under OSX 10.10.5

splaisan, 2016-07-29:
I installed 
sudo pip install snmp-passpersist
sudo port install libusb-devel
when I run sudo python setup.py install I get
<pre>
running install
running bdist_egg
running egg_info
writing requirements to temperusb.egg-info/requires.txt
writing temperusb.egg-info/PKG-INFO
writing top-level names to temperusb.egg-info/top_level.txt
writing dependency_links to temperusb.egg-info/dependency_links.txt
writing entry points to temperusb.egg-info/entry_points.txt
reading manifest file 'temperusb.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
writing manifest file 'temperusb.egg-info/SOURCES.txt'
installing library code to build/bdist.macosx-10.5-x86_64/egg
running install_lib
running build_py
creating build/bdist.macosx-10.5-x86_64/egg
creating build/bdist.macosx-10.5-x86_64/egg/temperusb
copying build/lib/temperusb/__init__.py -> build/bdist.macosx-10.5-x86_64/egg/temperusb
copying build/lib/temperusb/cli.py -> build/bdist.macosx-10.5-x86_64/egg/temperusb
copying build/lib/temperusb/snmp.py -> build/bdist.macosx-10.5-x86_64/egg/temperusb
copying build/lib/temperusb/temper.py -> build/bdist.macosx-10.5-x86_64/egg/temperusb
byte-compiling build/bdist.macosx-10.5-x86_64/egg/temperusb/__init__.py to __init__.pyc
byte-compiling build/bdist.macosx-10.5-x86_64/egg/temperusb/cli.py to cli.pyc
byte-compiling build/bdist.macosx-10.5-x86_64/egg/temperusb/snmp.py to snmp.pyc
byte-compiling build/bdist.macosx-10.5-x86_64/egg/temperusb/temper.py to temper.pyc
creating build/bdist.macosx-10.5-x86_64/egg/EGG-INFO
copying temperusb.egg-info/PKG-INFO -> build/bdist.macosx-10.5-x86_64/egg/EGG-INFO
copying temperusb.egg-info/SOURCES.txt -> build/bdist.macosx-10.5-x86_64/egg/EGG-INFO
copying temperusb.egg-info/dependency_links.txt -> build/bdist.macosx-10.5-x86_64/egg/EGG-INFO
copying temperusb.egg-info/entry_points.txt -> build/bdist.macosx-10.5-x86_64/egg/EGG-INFO
copying temperusb.egg-info/requires.txt -> build/bdist.macosx-10.5-x86_64/egg/EGG-INFO
copying temperusb.egg-info/top_level.txt -> build/bdist.macosx-10.5-x86_64/egg/EGG-INFO
zip_safe flag not set; analyzing archive contents...
creating 'dist/temperusb-1.5.1-py2.7.egg' and adding 'build/bdist.macosx-10.5-x86_64/egg' to it
removing 'build/bdist.macosx-10.5-x86_64/egg' (and everything under it)
Processing temperusb-1.5.1-py2.7.egg
Removing /opt/biotools/miniconda2/lib/python2.7/site-packages/temperusb-1.5.1-py2.7.egg
Copying temperusb-1.5.1-py2.7.egg to /opt/biotools/miniconda2/lib/python2.7/site-packages
temperusb 1.5.1 is already the active version in easy-install.pth
Installing temper-poll script to /opt/biotools/miniconda2/bin
Installing temper-snmp script to /opt/biotools/miniconda2/bin

Installed /opt/biotools/miniconda2/lib/python2.7/site-packages/temperusb-1.5.1-py2.7.egg
Processing dependencies for temperusb==1.5.1
Searching for pyusb==1.0.0
Best match: pyusb 1.0.0
Processing pyusb-1.0.0-py2.7.egg
pyusb 1.0.0 is already the active version in easy-install.pth

Using /opt/biotools/miniconda2/lib/python2.7/site-packages/pyusb-1.0.0-py2.7.egg
Finished processing dependencies for temperusb==1.5.1
</pre>

but when I try to launch '''temper-poll''' I get an error
<pre>
Traceback (most recent call last):
  File "/opt/biotools/miniconda2/bin/temper-poll", line 9, in <module>
    load_entry_point('temperusb==1.5.1', 'console_scripts', 'temper-poll')()
  File "build/bdist.macosx-10.5-x86_64/egg/temperusb/cli.py", line 36, in main
  File "build/bdist.macosx-10.5-x86_64/egg/temperusb/temper.py", line 340, in __init__
  File "build/bdist.macosx-10.5-x86_64/egg/usb/core.py", line 1263, in find
usb.core.NoBackendError: No backend available
</pre>

Any idea what I could do to improve this?
Thanks

padelt, 2016-07-29:
Hmm, I'd guess that the `libusb` backend that `pyusb` is using is missing, e.g. there is none or `pyusb` just cannot find it. [The FAQ](https://github.com/walac/pyusb/blob/master/docs/faq.rst) suggests that. In [the tutorial](https://github.com/walac/pyusb/blob/master/docs/tutorial.rst#specifying-libraries-by-hand) describes a way to specify a backend via the shared library path. Maybe you can find that using `find / -name "libusb*"`. Sorry, just guessing here. Never used it on a Mac.

padelt, 2017-03-31:
I guess you found a solution? Please re-open if not.

closed 2017-03-31 by padelt