padelt/temper-python #61: Temperature gets "stuck" or skips values

endolith, 2016-08-02:
So I've had this for years and it's always had this problem, but your "Reported working devices" includes this VID/PID?  So maybe there's a bug that's been fixed that you know about?

Here are the USB descriptors:

```
Bus 002 Device 002: ID 0c45:7401 Microdia TEMPer Temperature Sensor
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x0c45 Microdia
  idProduct          0x7401 TEMPer Temperature Sensor
  bcdDevice            0.01
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           59
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.10
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      65
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval              10
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.10
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      41
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval              10
```

and here is the log from several years:

[image]

As you can see, it goes up and down with the seasons, but avoids certain temperature ranges and gets "stuck" at certain temperatures.  If I do `diff(unique(data))`, though, it seems like it actually has 0.0625 °C resolution across the entire range:

[image]

It just strongly prefers certain temperatures over others.  I don't know if this is a bug in the hardware, firmware, or software that reads it, but I'm posting here in case you know.  It's been logging using cba5bf4d75f630610a440bfaeaaaf4a92e655079 since that date.  If you know this has nothing to do with your software, feel free to close it.

cooljimy84, 2017-03-20:
Got my TEMPer 1.4 and after a few polls it gets stuck at 22.7, even after rebooting it reports 22.7. I need to unplug it and wait a minute.
When the device works this project works perfectly

padelt, 2017-03-31:
@endolith If https://github.com/padelt/temper-python/issues/59#issuecomment-218011600 is correct then we should expect increments of 0.125°C as that would be the reported resultion. The "preferring" of some temperature I cannot explain, sorry. My guess would be an issue with the sensor or the way it works. This driver does not seem to be a likely cause.

padelt, 2017-03-31:
@cooljimy84 Your device (or this driver) reporting the same value over and over is likely a USB problem. We have had problems with the USB interactions initiated by this driver leading to "stuck" or resetting TEMPer devices. The root cause seems to be that we don't know for sure what we are doing in that regard. There are several different device revisions and the original source of the USB intercation was a more-or-less unmaintained C source. We have (or at least I have) no clear idea what exactly happens inside each TEMPer when this driver does its job. You could try to look in old issues for likely sources. But I guess we cannot help here, as it stands. sorry!

closed 2017-03-31 by padelt