padelt/temper-python #63: Misleading error message when run as non-root

eric-s-raymond, 2016-09-06:
Experimenting with repository head under Ubuntu 16.04 Linux on an Intel machine.

Device can be polled when running as root.  When running as non-root, temper-poll dies with the following stack trace:

esr@snark:~/software/temper-python$ temper-poll 
Traceback (most recent call last):
  File "/usr/local/bin/temper-poll", line 9, in <module>
    load_entry_point('temperusb==1.5.1', 'console_scripts', 'temper-poll')()
  File "build/bdist.linux-x86_64/egg/temperusb/cli.py", line 36, in main
  File "build/bdist.linux-x86_64/egg/temperusb/temper.py", line 340, in __init__
  File "build/bdist.linux-x86_64/egg/temperusb/temper.py", line 89, in __init__
  File "build/bdist.linux-x86_64/egg/temperusb/temper.py", line 140, in lookup_sensor_count
  File "build/bdist.linux-x86_64/egg/usb/core.py", line 841, in product
  File "build/bdist.linux-x86_64/egg/usb/util.py", line 314, in get_string
ValueError: The device has no langid

The following comment thread suggests a possible solution:

https://github.com/walac/pyusb/issues/139

padelt, 2016-09-07:
I can reproduce this if the current user has no write permission to the USB device node:
```
(v1)pa@plug2:~/temper/padelt$ lsusb | grep Microdia
Bus 001 Device 004: ID 0c45:7401 Microdia
Bus 001 Device 005: ID 0c45:7401 Microdia
(v1)pa@plug2:~/temper/padelt$ ll /dev/bus/usb/001/00[45]
crw-rw-r-T 1 root root 189, 3 Sep  7 09:04 /dev/bus/usb/001/004
crw-rw-r-T 1 root root    189, 4 Sep  7 09:04 /dev/bus/usb/001/005
(v1)pa@plug2:~/temper/padelt$ groups
pa root
(v1)pa@plug2:~/temper/padelt$ temper-poll
Found 2 devices
Device #0: 33.7°C 92.6°F
Device #1: 21.4°C 70.6°F
(v1)pa@plug2:~/temper/padelt$ sudo chmod g-w /dev/bus/usb/001/004
(v1)pa@plug2:~/temper/padelt$ temper-poll
Traceback (most recent call last):
  File "/home/pa/temper/padelt/v1/bin/temper-poll", line 9, in <module>
    load_entry_point('temperusb==1.5.1', 'console_scripts', 'temper-poll')()
  File "build/bdist.linux-armv7l/egg/temperusb/cli.py", line 36, in main
  File "build/bdist.linux-armv7l/egg/temperusb/temper.py", line 340, in __init__
  File "build/bdist.linux-armv7l/egg/temperusb/temper.py", line 89, in __init__
  File "build/bdist.linux-armv7l/egg/temperusb/temper.py", line 140, in lookup_sensor_count
  File "build/bdist.linux-armv7l/egg/usb/core.py", line 841, in product
  File "build/bdist.linux-armv7l/egg/usb/util.py", line 314, in get_string
ValueError: The device has no langid
```
Fix coming soon.

padelt, 2016-09-07:
I just published the code to improve the error message. The changes are not yet available on PyPI, but if you `git pull` and `python setup.py install` again you should get a more comprehensible one.
Thanks for reporting the issue!

closed 2016-09-07 by padelt