padelt/temper-python #64: Update PyPI with version 1.5.2

padelt, 2016-09-07:
@amorphic Could you please upload the new version 1.5.2 to PyPI?

padelt, 2016-09-07:
BTW: Is there a way for me to publish to PyPI, too? It's great that you are doing that but I don't want to hassle you too much.

padelt, 2017-03-31:
@amorphic A new release is coming around soon. Could you allow me to publish to PyPI somehow?

amorphic, 2017-04-01:
@padelt so sorry I completely missed this the first time around! Yes I'm sure I'll be able to give you admin rights. First thing is probably to create a PyPI account if you haven't done so yet.

padelt, 2017-04-01:
Hi James, no worries! I just registered a user at PyPI: padelt
Could you try it with that?
Philipp

amorphic, 2017-04-03:
Ok Philipp, I've just added you with the __Owner__ role. Let me know if it works.

padelt, 2017-04-03:
Thanks James, that worked!

closed 2017-04-03 by padelt