padelt/temper-python #66: Incorrect result when temperature drops below freezing

alanwbaker, 2016-12-07:
I'm using a 0c45:7401 Microdia device.  temper-poll 1.5.2 works great above freezing, but when the temperature drops below 32F/0C, it returns 498 as the Fahrenheit temperature and 254 as the Centigrade temperature.  Am I missing something?

Alan

fiz1962, 2016-12-09:
The problem is at line 308 in temper.py.  The data[offset] and data[offset+1] values go with temperature (in Celcius) as
temp  data[offset]  data[offset+1]
0.2      0                  64
0.1      0                  32
0.0      0                  0
-0.1    255              224
-0.2    255              192

Ie, when the temp drops below 0C, 256 needs to be subtracted from the temperature like

            celsius = data[offset] + data[offset+1] / 256.0
	    if data[offset] > 127:
		celsius = celsius - 256
            celsius = celsius * self._scale + self._offset

smeek, 2017-01-09:
I also ran into this and have an open pull request (https://github.com/padelt/temper-python/pull/68) to fix it.

martin3000, 2017-01-26:
Hmm...for years nobody saw this?
            celsius=struct.unpack_from('>h',data,offset)[0]/256.0

padelt, 2017-03-31:
This should be fixed in the new master revision. If the issue re-appears, please open the issue again.

closed 2017-03-31 by padelt