padelt/temper-python #69: Adding support for Relative Humidity

smeek, 2017-01-09:
I have a TEMPerHUM device (USB ID 0c45:7402, TEMPer1F_H1_V1.4) which I have working nicely with the current code after some small tweaks (pull requests already in).

I'd like to add support for reading the relative humidity, but before embarking on anything wondered what your preference was for adding this?

Currently the code is structured around the fact that you'll only be reading temperature, which means to add it cleanly it might require a sizable re-work.

Having sniffed the USB traffic and compared the values returned against the output of the Windows program, I know the conversion required:
`rel_humidity = (((data[offset+2] << 8) + data[offset+3]) * 32) / 1000.0`
I have hacked it in locally (immediately after reading the temperature in `get_temperatures`) to show it works.

Any comments on how you would like this structured / approached?

padelt, 2017-01-09:
I think we could get away with changing `get_data()` to return something like `{'temp_data': ..., 'humidity_data': ...}` and add another method like `get_humidity()` that calls `get_data()` too. That would mean in a normal `cli.py` run, `get_data()` would get called twice. Inefficient but not that bad.

In the long run, all special casing for different devices should go into device-specific classes. As part of that I would split the request cycle into getting fresh data via USB and intepreting the (possibly cached) data. A saner approach to multi-sensor modules would follow too.

smeek, 2017-01-10:
Yes, inefficient, but I easier to extend for other humidity sensing devices in the future.
Just because the temperature and humidity data comes back in the same response to the one `temp` USB command for this device, doesn't necessarily mean that will always be the case for other/future devices.
I'll try to put something together along the lines you suggest when I get some time...

padelt, 2017-01-10:
Great, looking forward to it!

smeek, 2017-01-20:
OK, I have created a pull request with my changes (https://github.com/padelt/temper-python/pull/70).
Basically as you suggested (I hope).
Slightly nasty to combine the temperature and humidity readings into one list per device.
Seems to work, although I do not have any other devices to test against; I just had to comment out the bits of code that produced the humidity readings to make sure that the code ignored it if it couldn't find `humidity_pc`.

padelt, 2017-01-20:
I didn't have a chance to look yet but what comes to mind: Can I buy one of the devices you have? Do you have a link? I have several others and testing multiple devices would be really helpful.

smeek, 2017-01-20:
I am in the UK and just got it from Amazon (https://www.amazon.co.uk/TEMPerHUM-Sensor-USB-Hygrometer-Thermometer/dp/B00K6DN7HE/ref=sr_1_1?ie=UTF8&qid=1484933713&sr=8-1&keywords=TEMPerHUM).
Although the device I received was in a white plastic case, rather than the metal case that was shown in the pictures (not sure if that makes a difference as to what the device announces itself as).

padelt, 2017-03-31:
Just merged it - please reopen if needed!

closed 2017-03-31 by padelt