padelt/temper-python #73: Temper-poll timeouts then stops getting data

Pizzicato, 2017-03-13:
Hi,
Thanks for this software :-)
I'm trying to make it work, but I keep getting this output after running the command for some time, once every 30 second. After an hour or so it starts crashing with this message:
```
ERROR:temperusb.temper:[Errno 110] Operation timed out
Traceback (most recent call last):
  File "/usr/bin/temper-poll", line 9, in <module>
    load_entry_point('temperusb==1.5.2', 'console_scripts', 'temper-poll')()
  File "/usr/lib/python2.6/site-packages/temperusb-1.5.2-py2.6.egg/temperusb/cli.py", line 53, in main
    readings.append(dev.get_temperatures(sensors=sensors))
  File "/usr/lib/python2.6/site-packages/temperusb-1.5.2-py2.6.egg/temperusb/temper.py", line 300, in get_temperatures
    data = self.get_data()
  File "/usr/lib/python2.6/site-packages/temperusb-1.5.2-py2.6.egg/temperusb/temper.py", line 250, in get_data
    return self.get_data(True)
  File "/usr/lib/python2.6/site-packages/temperusb-1.5.2-py2.6.egg/temperusb/temper.py", line 230, in get_data
    self._interrupt_read()
  File "/usr/lib/python2.6/site-packages/temperusb-1.5.2-py2.6.egg/temperusb/temper.py", line 334, in _interrupt_read
    data = self._device.read(ENDPOINT, REQ_INT_LEN, timeout=TIMEOUT)
  File "/usr/lib/python2.6/site-packages/pyusb-1.0.0-py2.6.egg/usb/core.py", line 988, in read
    self.__get_timeout(timeout))
  File "/usr/lib/python2.6/site-packages/pyusb-1.0.0-py2.6.egg/usb/backend/libusb1.py", line 851, in intr_read
    timeout)
  File "/usr/lib/python2.6/site-packages/pyusb-1.0.0-py2.6.egg/usb/backend/libusb1.py", line 936, in __read
    _check(retval)
  File "/usr/lib/python2.6/site-packages/pyusb-1.0.0-py2.6.egg/usb/backend/libusb1.py", line 595, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 110] Operation timed out
```
Once this happens, it happens all the time.

I get this on /var/log/messages (or dmesg):
```
Mar 13 14:19:16 master kernel: usb 4-2: reset low speed USB device number 66 using ohci_hcd
Mar 13 14:19:16 master kernel: usb 4-2: usbfs: process 61958 (temper-poll) did not claim interface 1 before use
Mar 13 14:19:16 master kernel: usb 4-2: reset low speed USB device number 66 using ohci_hcd
```

The last line repeats itself after each execution of the command.

This is the device information (lsusb -v)
```
Bus 004 Device 066: ID 0c45:7401 Microdia 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x0c45 Microdia
  idProduct          0x7401 
  bcdDevice            0.01
  iManufacturer           1 RDing
  iProduct                2 TEMPerV1.4
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           59
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
[... 128 lines cut ...]
            Item(Global): Report Size, data= [ 0x08 ] 8
            Item(Global): Report Count, data= [ 0x08 ] 8
            Item(Main  ): Feature, data= [ 0x02 ] 2
                            Data Variable Absolute No_Wrap Linear
                            Preferred_State No_Null_Position Non_Volatile Bitfield
            Item(Main  ): End Collection, data=none
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval              10
Device Status:     0x0000
  (Bus Powered)
```

I'm using RHEL6.

I've read some similar issues, but I think they were marked as solved after changing the code. My temper-poll version is the last one, I just reinstalled it to be sure in included the last updates.

Cheers

Pizzicato, 2017-03-23:
Hi,
I have been using a C program based on [this one](https://github.com/padelt/pcsensor-temper/blob/master/src/pcsensor.c), modified to be able to use it with up to 4 devices, since it was made for one.

When I use this program, instead of crashing every-time after it times-out for the first time, it works when being run again. [Here](https://gist.github.com/Pizzicato/d6519a665d71a309b99d648b545c2da5) is the gist of my version, in case you can get any clue of what's going on with temper-python. When using this program, lsusb shows that the device number is increased by one after each crash ¿?

Cheers

padelt, 2017-03-31:
Hi @Pizzicato , thanks for the info! I think you are seeing a problem similar to what I described here:
https://github.com/padelt/temper-python/issues/61#issuecomment-290753783
There most likely is something in the USB interaction with your device that makes it choke and reset. That would explain the increasing device number.
Problem is: I have neither the device nor the time to look at this.
What you could do is trying to carefully inspect what the C version does (or doesn't) and compare it to the code around here:
https://github.com/padelt/temper-python/blob/master/temperusb/temper.py#L239
Then try to mimick what the C version does in `temper.py`.
Maybe you can find a combination that works for you! If you do, please report it so we can add it to the codebase.
Until then, I'll close this issue.

closed 2017-03-31 by padelt

Pizzicato, 2017-04-05:
Hi!
I'm quite busy right now too, but I'll try to do it and fix it.

Cheers

Pizzicato, 2017-04-19:
Hi,
I have found out a couple of things that might be useful: temper-poll device resetting doesn't actually resets my USB devices. Using [this C program](https://gist.github.com/Pizzicato/d6519a665d71a309b99d648b545c2da5), which is a little modification of the one I posted before (adds a call to `usb_reset` when reading fails) changes the failing device ID, which can be seen using lsusb, while temper-poll doesn't.

So when the USB device fails, if reset is done properly, subsequent reads should work. I don't have time to make the changes in the temper-poll script, but this may point someone out in the right direction to fix this issue.

Cheers