padelt/temper-python #78: Raspberry pi

santiagomoneta, 2017-09-29:
Hi, im testing your code on my raspberry pi and im facing some issues... the 1st one was that the 10th line of the setup, the word "encoding" was causing problems:

`raceback (most recent call last):
  File "setup.py", line 10, in <module>
    long_description=open('README.md', encoding='utf-8').read(),
TypeError: 'encoding' is an invalid keyword argument for this function`

I commented that line and was able to install, but when I run the temper-poll command I get this:

`Traceback (most recent call last):
  File "/usr/local/bin/temper-poll", line 9, in <module>
    load_entry_point('temperusb==1.5.3', 'console_scripts', 'temper-poll')()
  File "build/bdist.linux-armv7l/egg/temperusb/cli.py", line 38, in main
  File "build/bdist.linux-armv7l/egg/temperusb/temper.py", line 420, in __init__
  File "build/bdist.linux-armv7l/egg/temperusb/temper.py", line 102, in __init__
usb.core.USBError: [Errno None] Error reading langids from device. This might be a permission issue. Please check that the device node for your TEMPer devices can be read and written by the user running this code. The temperusb README.md contains hints about how to fix this. Search for 'USB device permissions'.`

I went over the permission on the USB dongle and while I found it here:

`pi@BA6:/dev $ lsusb | grep "0c45:7401"
Bus 001 Device 103: ID 0c45:7401 Microdia`

I don't find it here:

`pi@BA6:/dev $ ls -l /dev/usb*
ls: cannot access /dev/usb*: No such file or directory`

I tried to run the commands on sudo but is still not working... any ideas?

Thanks.

santiagomoneta, 2017-09-29:
I add this:

input: RDing TEMPerV1.4 as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.5/1-1.5:1.0/0003:0C45:7401.00C5/input/input98
[499738.463052] hid-generic 0003:0C45:7401.00C5: input,hidraw0: USB HID v1.10 Keyboard [RDing TEMPerV1.4] on usb-3f980000.usb-1.5/input0
[499738.470908] hid-generic 0003:0C45:7401.00C6: hiddev0,hidraw1: USB HID v1.10 Device [RDing TEMPerV1.4] on usb-3f980000.usb-1.5/input1
[504838.247426] usb 1-1.5: USB disconnect, device number 102
[504838.523455] usb 1-1.5: new low-speed USB device number 103 using dwc_otg
[504838.641482] usb 1-1.5: New USB device found, idVendor=0c45, idProduct=7401
[504838.641501] usb 1-1.5: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[504838.641513] usb 1-1.5: Product: TEMPerV1.4
[504838.641523] usb 1-1.5: Manufacturer: RDing

Stone-Hedge, 2017-10-09:
I'm seeing this problem also, on latest raspbian-lite.

Stone-Hedge, 2017-10-09:
I think I have a solution. You need to run with Python3, and make sure you execute as root afterwards.

To do this, install python3 setuptools:

`sudo apt-get install python3-setuptools`

execute the installation script with:

`sudo python3 setup.py install`

and run temper-poll as root

`sudo temper-poll`

Worked for me

Stone-Hedge, 2017-10-09:
I think this might only be a workaround as temper-snmp doesn't work with python3.  Works fine if you don't need SNMP though

stethewwolf, 2018-01-10:
Hi, I have tried to use, successfully, a virtual env both with python 2 and with python3.

rpi3 with Fedora release 26 (Twenty Six)

LaszloGombos, 2019-01-04:
Create a new file for udev (e.g. /etc/udev/rules.d/99-tempsensor.rules) with the following 2 lines
SUBSYSTEMS=="usb", ACTION=="add", ATTRS{idVendor}=="0c45", ATTRS{idProduct}=="7401", MODE="666"
SUBSYSTEMS=="usb", ACTION=="add", ATTRS{idVendor}=="0c45", ATTRS{idProduct}=="7402", MODE="666"

Save the file unplug and plug back the device and try again as a user (instead of root). Let us know if this helped !

BrianGilbert, 2021-05-13:
> I think I have a solution. You need to run with Python3, and make sure you execute as root afterwards.
>
> To do this, install python3 setuptools:
>
> `sudo apt-get install python3-setuptools`
>
> execute [...]

I did the same when I hit issues and it's working for me too.

padelt, 2021-05-14:
I'll close this issue as I suspect it is irrelevant by now. Please reopen as needed.

closed 2021-05-14 by padelt