padelt/temper-python #79: Temperature is never below -1.8 C

ilkkao, 2017-10-21:
`dmesg` output: `[    2.225797] usb 1-1.5: Product: TEMPer2_M12_V1.3`

Command I use:

```
$ raspberrypi  ~/g/temper-python> temper-poll -s 1                                                                                                                                                    
Found 1 devices
Device #0: -1.8°C 28.7°F
```

Result in Finland currently:

[image]

During the daytime, temperature is correctly reported. But at night when it gets freezing, values never go below `-1.8C`. In reality, it gets colder than that. 

My temper device should work up to `-40C`. Could it be that something changes at `-1.8C` regarding how the value is read? I'm also wondering if my device is faulty but that doesn't feel likely as day temperatures are very accurate.

ilkkao, 2017-10-21:
I added `print` to see the raw response byte array:

```
 raspberrypi  ~/g/temper-python> /home/pi/.local/bin/temper-poll -s 1                                                                                                                               master!
Found 1 devices
{'temp_data': array('B', [128, 4, 5, 128, 254, 48, 191, 107]), 'humidity_data': None}
Device #0: -1.8°C 28.7°F
 raspberrypi  ~/g/temper-python> /home/pi/.local/bin/temper-poll -s 1                                                                                                                               Found 1 devices
{'temp_data': array('B', [128, 4, 5, 144, 254, 48, 191, 107]), 'humidity_data': None}
Device #0: -1.8°C 28.7°F
```

The fourth byte changed but it didn't affect the reported value. If I read the source code correctly (never really used Python) the temperature from the second sensor is read solely from the fifth byte. Maybe the fourth byte affects it too? Especially as the fifth byte has almost maximum value.

Are there any kind of manufacturer specs available?

ilkkao, 2017-10-21:
I see now that changed value in my example is from the indoor sensor and sensor value is 16-bit integer.

kl1lercher, 2017-11-26:
Same here for me, Temperature is never below -1.6°C. As i have two sensors i've changed positions (inside/outside a vice versa) but still temperature is never below -1.6°C. I am still checking data output to try to find out if it's a hardware problem (TEMPerV1.4)

ilkkao, 2017-11-27:
My sensor started to work again when temperature went below ~ -4C (can't remember the exact value, the sensor is offline now). 

Not sure at all what the problem is at this point.

ilkkao, 2018-02-22:
Winter came so I was able to debug this bit further. My sensor return -1.8C when the temperature is between 0..-1.8C. It works just fine when the temperature is outside of this range, i.e >0 or <1.8.

A strange HW flaw is my guess. Feel free to close this issue.

closed 2021-05-14 by padelt