padelt/temper-python #84: Support for product ID 413d:2107

kenkma, 2018-01-25:
Most recent sensors seems to have a new ID, 413d:2107.

Has anyone successfully hacked the code to work with it?  

this C hack seems to work.

https://github.com/edorfaus/TEMPered/issues/51

Thank you.

Estebastien, 2018-01-26:
Yes same need. I have a TEMPerX232 (thermo-hygrometer) with that ID and I can't make it work as my others Temper1F.
I have no clue. I don't really now what magic is going on in these hacks. I added the new ID to the VIPDS list, and of course I get an error ("Entity not found").
Thank you

mreymann, 2018-02-07:
Hi I use temper-python with my Microdia TEMPer Temperature Sensor (0c45:7401) and it works like a charm. I, as well, was too adventurous when buying an obscure "TEMPerHUM" (413d:2107). It would be great to have support for this device in temper-python!

Thanks for your work!

mreymann, 2018-02-16:
I've created a working Munin plugin for our device: https://github.com/mreymann/temperx
Support should be fairly easy to implement in temper-python. USB commands seem to be very similar. I have a PCAP file here if anybody needs it.

ndemou, 2018-06-01:
Hoping that this comment will help temper-python to support devices with id 413d:2107 I should note that running this hid-query  gives you the temperature (hid-query is part of https://github.com/edorfaus/TEMPered more info bellow):
```
./hid-query /dev/hidraw1 0x01 0x80 0x33 0x01 0x00 0x00 0x00 0x00
...
Response from device (8 bytes):
         80 80 09 0e   4e 20 00 00     <--- temperature is 0x090e/100
```
-----

P.S. Until temper-python supports them here's a complete set of commands to print the temperature:
```
OUTLINE=`./hid-query /dev/hidraw1 0x01 0x80 0x33 0x01 0x00 0x00 0x00 0x00|grep -A1 ^Response|tail -1`
OUTNUM=`echo $OUTLINE|sed -e 's/^[^0-9a-f]*[0-9a-f][0-9a-f] [0-9a-f][0-9a-f] \([0-9a-f][0-9a-f]\) \([0-9a-f][0-9a-f]\) .*$/0x\1\2/'`
printf "%.1f°C\n" $(( $OUTNUM/100 ))

23.0°C
```

The above are 99% based on TEMPered#51 "hack" from digiboule mentioned above.

To install TEMPered (that contains the hid-query utility) download it and see this post https://github.com/edorfaus/TEMPered/issues/35#issuecomment-128130774

P.S.3:  I must admit that I'm really confused by who forked what version of TEMPered to support which device type. 

P.S.4: Here's a copy of my bash session that ended up on the above commands:

```
root@lion:~/manual-install/TEMPered-master/utils#  lsusb
...
Bus 004 Device 002: ID 413d:2107  
...
root@lion:~/manual-install/TEMPered-master/utils# ./hid-query --enum
/dev/hidraw0 : 413d:2107 interface 0 : (null) (null)
/dev/hidraw1 : 413d:2107 interface 1 : (null) (null)

root@lion:~/manual-install/TEMPered-master.TafThorne/build/utils# ./hid-query /dev/hidraw0 0x01 0x80 0x33 0x01 0x00 0x00 0x00 0x00
Device /dev/hidraw0 : 413d:2107 interface 0 : (null) (null)

Writing data (9 bytes):
         00 01 80 33   01 00 00 00   00

No data was read from the device (timeout).

root@lion:~/manual-install/TEMPered-master/utils# ./hid-query /dev/hidraw1 0x01 0x80 0x33 0x01 0x00 0x00 0x00 0x00
Device /dev/hidraw1 : 413d:2107 interface 1 : (null) (null)

Writing data (9 bytes):
         00 01 80 33   01 00 00 00   00

Response from device (8 bytes):
         80 80 09 0e   4e 20 00 00
```

jbeale1, 2018-06-17:
As I mentioned at https://github.com/edorfaus/TEMPered/issues/35#issuecomment-128130774  I have one of these new 413d:2107 devices which the Windows "TEMPer V26.1" software calls "Firmware: TEMPerGold_V3.1"  and mine is I think temperature only, although the hidapi testgui app shows two devices with that VID:PID.  The eight-byte hex string posted by ndemou above, does work when manually entered from the hidapi testgui.  However the TEMPered project "hid-query" tool does not see the device at all (only sees my keyboard and mouse) so I don't know how to script this.

ndemou, 2018-06-17:
@jbeale1:  mine is also temperature only and if you notice in the transcript above I also see two devices and one of them doesn't respond.

jbeale1, 2018-06-17:
It does work! For me it's hidraw3 and I think before I wasn't root and I didn't do 'sudo' 
```
 sudo ./hid-query /dev/hidraw3 0x01 0x80 0x33 0x01 0x00 0x00 0x00 0x00
Device /dev/hidraw3 : 413d:2107 interface 1 : (null) (null)
Writing data (9 bytes):  00 01 80 33   01 00 00 00   00
Response from device (8 bytes):  80 80 08 4d   4e 20 00 00
```

jbeale1, 2018-06-17:
Bash script to print time & temp every ~15 sec:
```
while [ true ]; do
  temp=`/usr/local/bin/readTEMPer.sh`
  echo $(date +"%F %T") " , " $temp
  sleep 14
done
```
example output:
```
2018-06-17 11:45:23  ,  24.50
2018-06-17 11:45:38  ,  24.43
2018-06-17 11:45:52  ,  24.25
2018-06-17 11:46:07  ,  24.12
2018-06-17 11:46:21  ,  23.93
2018-06-17 11:46:36  ,  23.43
2018-06-17 11:46:50  ,  23.43
2018-06-17 11:47:05  ,  23.43
```

ndemou, 2018-06-17:
Thanks @jbeale1 for noticing the bug with integer arithmetic. I fixed my script (just used awk instead of bc because which works perfectly for such low precision calculations and is pre-installed almost everywhere)

jbeale1, 2018-06-17:
I wonder if I have a different version of 'awk' because it always gives me 0.0 for example:
```
awk "BEGIN {printf(\"%4.1f\n\",0x0891/100)}"
 0.0
```

ndemou, 2018-06-17:
Right again. A cross platform way send to be echo $(( 16#891 ))|awk '{printf("%4.1f\n",$0/100)}'

schniepp, 2018-06-19:
My biggest issue right now is that `hid-query /dev/hidraw3 0x01 0x86 0xff 0x01 0x00 0x00 0x00 0x00` gives me this error: `Write failed: (null)`. I am doing this as root, and hid-query detects it successfully. Is this perhaps a slightly different device? Does it need a different string sent? Or is something about permissions set up the wrong way, so that writing does not work?
This device is supposed to do both T and R.H.: [vendor](http://www.pcsensor.com/usb-temperature-humidity/usb-hygrometers/temperx232.html), [Amazon](https://www.amazon.com/Temperature-Humidity-Secondary-Development-TEMPerX232/dp/B076MZVDVK)

ndemou, 2018-06-19:
@schniepp  have you tried all devices that `hid-query --enum` returns?

schniepp, 2018-06-19:
Yes. The other two ones are only my external keyboard and mouse. The one that I am using reports the correct USB ID.

jbeale1, 2018-06-19:
After logging for two days, I see the output of my TEMPer has a nominal 0.06 C resolution. I haven't done a careful calibration, but the plot shows big flat spots at fixed values like it gets stuck. It is not, but clearly it can not reliably measure changes smaller than 0.5 C. Example plot:  https://photos.app.goo.gl/HEiZdxNJTLN1RLUf7

cmsimike, 2018-07-06:
I received a 413d:2107 from amazon, and though it reports two USB devices (keyboard/USB device), I can only seem to get data from the usb device. The temp it reports back is way too high to make sense. It was commented elsewhere that there is an internal and external temp that can be reported. I'm worried that I'm only seeing the internal temp, if that's the case.

Proxx, 2018-08-21:
same here, got the 413d:2107 version and really like to see this usbstick supported 😄

cmsimike, 2018-08-21:
I am currently using this script to read the raw USB values from the 413d:2107

```
#!/bin/bash
OUTLINE=`/root/dev/TEMPered/build/utils/hid-query /dev/hidraw3 0x01 0x80 0x33 0x01 0x00 0x00 0x00 0x00|grep -A1 ^Response|tail -1`
OUTNUM=`echo $OUTLINE|sed -e 's/^[^0-9a-f]*[0-9a-f][0-9a-f] [0-9a-f][0-9a-f] \([0-9a-f][0-9a-f]\) \([0-9a-f][0-9a-f]\) .*$/0x\1\2/'`
HEX4=${OUTNUM:2:4}
DVAL=$(( 16#$HEX4 ))
CTEMP=$(bc <<< "scale=2; $DVAL/100")
FTEMP=$(echo "scale=2;((9/5) * $CTEMP) + 32" |bc)
```
I build TEMPered, and it has an hid-query, which I can use to get the raw sensor values.

schniepp, 2018-10-16:
This totally fixed my problems: [urwen/temper](https://github.com/urwen/temper).
It's a pure python3 program that makes my device work. The author also describes in the README.md that the access via hidraw does not work and gives errors. Brilliant! I can use my device now!!!

poikilos, 2021-01-21:
> This totally fixed my problems: [urwen/temper](https://github.com/urwen/temper).
> It's a pure python3 program that makes my device work. The author also describes in the README.md that the access via hidraw does not work and gives errors. Brilliant! I can use my device now!!! The author also has listed a table with different devices.
> 
> What brought me on this track: the Firmware version using the Windows software reads as "TEMPerX232_V2.0".

A fork of it has several improvements. It also has a setup.py so it can be installed (in a `virtualenv` preferably) like:
```
python -m pip install --upgrade pip
python -m pip install --upgrade setuptools wheel

```

acs4design, 2022-10-21:
I was able to get my Temper2 device working in windows. This device is VID 0x431D PID 0x2107.  There is no product name in the USB descriptor, so there's no good way to tell it apart from any of the others except for VID/PID. It does have 2 endpoints.  The first is a keyboard endpoint. The other is the temperature interface.  This device has a button on it.  It has an internal temp measurement and an external measurement probe.  To get both sensor readings, you write the USB commands for init, then send USB command to read temp, then complete 2 reads.  The first read is the internal temp (offset 2 in the 8 byte data) then divide by 100.0.  The second read is the external temp (offset 2 in the 8 byte data) then divide by 100.0.

davet2001, 2022-10-27:
@acs4design Thanks for sharing this.  Is your sensor like one of the ones here?:

https://usb-ids.gowdy.us/read/UD/413d/2107

Please could you double check the description strings to see if there's anything uniquely identifiable about the sensor that we can use to select the correct decoding?

I don't have one of these sensors so I'm also looking for someone who can help me with testing a patch for this new sensor.

acs4design, 2022-10-27:
I can't find anything in the descriptor itself.  Perhaps, there are other
methods to determine a way to discriminate it from the others by receiving
the output reports. Their software somehow knows which Temper device it is
because their software supports all of their USB sticks. I used the
application that came with the stick to get it running while I did a USB
capture.   I used USBlyzer software to capture the USB data while it was
recording temperatures.  You can open the .ulz file with that software and
see the captured data.

I have attached the complete Device Descriptor (.html), USB packet
capture(.ulz) and pictures :)

Best Regards,
Jason Gurley

-- 
==========================

This e-mail, including any attachments, is 
intended for the exclusive use of the person(s) to which it is addressed 
and may contain proprietary, confidential and/or privileged information. If 
the reader of this e-mail is not the intended recipient or his or her 
authorized agent, any review, use, printing, copying, disclosure, 
dissemination or distribution of this e-mail is strictly prohibited. If you 
think that you have received the e-mail in error, please notify the sender 
immediately by return e-mail, delete this communication and destroy all 
copies.

==========================