padelt/temper-python #86: 3 sensor support

rtaft, 2018-06-30:
I have a TEMPerNTC II
10K F3950 NTCsensor

I noticed this line in lookup_sensor_count
# All others are two - if not the case, contribute here: https://github.com/padelt/temper-python/issues

Bus 003 Device 014: ID 0c45:7401 Microdia TEMPer Temperature Sensor

[ 3188.367137] usb 3-10.3: new low-speed USB device number 16 using xhci_hcd
[ 3188.448856] usb 3-10.3: New USB device found, idVendor=0c45, idProduct=7401
[ 3188.448859] usb 3-10.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 3188.448860] usb 3-10.3: Product: TEMPerNTC1.O
[ 3188.448862] usb 3-10.3: Manufacturer: RDing
[ 3188.452222] input: RDing TEMPerNTC1.O as /devices/pci0000:00/0000:00:14.0/usb3/3-10/3-10.3/3-10.3:1.0/0003:0C45:7401.0008/input/input21

Yes, that really says 1.O as in the letter O.

It has an internal sensor and 2 external sensors.  temper-python works fine for the internal and first external, but it won't read the 3rd without some code changes.  After adding 'TEMPerNTC1.O' with a count of 3 to temper.py, it works fine with all 3.

padelt, 2020-01-25:
Hey @rtaft , could you put the changes in a PR that I can merge?

rtaft, 2020-01-27:
@padelt done

padelt, 2020-01-27:
and merged!

closed 2020-01-27 by padelt