padelt/temper-python #87: fixed the line 10 error as in #80 but resulting exe does not work

splaisan, 2018-07-16:
It used to run under previsou OSX but not now under high sierra
Any idea what I am missing?
A dependency I guess!!

```
temper-poll -p
Traceback (most recent call last):
  File "/opt/miniconda2/bin/temper-poll", line 11, in <module>
    load_entry_point('temperusb==1.5.3', 'console_scripts', 'temper-poll')()
  File "/opt/biotools/temper-python/temperusb/cli.py", line 38, in main
    th = TemperHandler()
  File "/opt/biotools/temper-python/temperusb/temper.py", line 420, in __init__
    usb.core.find(find_all=True, idVendor=vid, idProduct=pid)]
  File "build/bdist.macosx-10.6-x86_64/egg/usb/core.py", line 1263, in find
usb.core.NoBackendError: No backend available
```

splaisan, 2018-07-20:
same issue after installing python3 and reinstalling temper under it (line10 edited back to repo code)

```
temper-poll -p
Traceback (most recent call last):
  File "/opt/miniconda2/envs/python3/bin/temper-poll", line 11, in <module>
    load_entry_point('temperusb==1.5.3', 'console_scripts', 'temper-poll')()
  File "/opt/miniconda2/envs/python3/lib/python3.6/site-packages/temperusb/cli.py", line 38, in main
    th = TemperHandler()
  File "/opt/miniconda2/envs/python3/lib/python3.6/site-packages/temperusb/temper.py", line 420, in __init__
    usb.core.find(find_all=True, idVendor=vid, idProduct=pid)]
  File "/opt/miniconda2/envs/python3/lib/python3.6/site-packages/usb/core.py", line 1263, in find
    raise NoBackendError('No backend available')
usb.core.NoBackendError: No backend available
```

padelt, 2021-05-14:
Closing because of age. Not sure we can help much here. `pyusb` is trying to find a backend (operating system level library) for USB and not finding any. Please reopen as needed.

closed 2021-05-14 by padelt