padelt/temper-python #89: upowerd spamming syslog when polling device

rdica, 2018-11-30:
Is anyone else getting syslog spam when polling your sensor?

Nov 29 19:47:01 daw3 journal[2761]: unhandled action 'bind' on /sys/devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.1
Nov 29 19:47:01 daw3 journal[2761]: unhandled action 'unbind' on /sys/devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.1
Nov 29 19:47:01 daw3 journal[2761]: unhandled action 'bind' on /sys/devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.1
Nov 29 19:47:01 daw3 journal[2761]: unhandled action 'unbind' on /sys/devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.1
Nov 29 19:48:01 daw3 journal[2761]: unhandled action 'bind' on /sys/devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.1
Nov 29 19:48:01 daw3 journal[2761]: unhandled action 'unbind' on /sys/devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.1
Nov 29 19:48:01 daw3 journal[2761]: unhandled action 'bind' on /sys/devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.1
Nov 29 19:48:01 daw3 journal[2761]: unhandled action 'unbind' on /sys/devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.1

From what I can find, it's upowerd reporting uevents in newer kernels per https://bugs.freedesktop.org/show_bug.cgi?id=106019
but I haven't found a means to kill the syslog spam. Using Fedora 28 and kernel 4.19.3-200.fc28.x86_64, upower-0.99.7-3.fc28.x86_64.

padelt, 2020-01-25:
I‘d be really interested in knowing the reason for this. It may be that temperusb is doing something questionable regarding binding. It anyone has a good explanation or even a suggestion what to improve, that would help.
As it stands, I have no idea how to improve it. Or if it needs improving at all.

rdica, 2020-11-28:
While I never did find out why, its moot now since F28 is EOL and I don't see the problem on F32 currently. Closing.

closed 2020-11-28 by rdica