padelt/temper-python #103: syntax error on debian buster

ReachableCEO, 2021-02-26:
root@pfv-vmsrv-03:~# temper-poll 
Traceback (most recent call last):
  File "/usr/local/bin/temper-poll", line 11, in <module>
    load_entry_point('temperusb==1.5.3', 'console_scripts', 'temper-poll')()
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 489, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2793, in load_entry_point
    return ep.load()
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2411, in load
    return self.resolve()
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2417, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "build/bdist.linux-x86_64/egg/temperusb/__init__.py", line 1, in <module>
  File "build/bdist.linux-x86_64/egg/temperusb/temper.py", line 17, in <module>
  File "/usr/local/lib/python2.7/dist-packages/temperusb-1.5.3-py2.7.egg/temperusb/device_library.py", line 21
    temp_sens_offsets: list,
                     ^
SyntaxError: invalid syntax

ReachableCEO, 2021-02-26:
This is from a git clone about 10 minutes ago or so

padelt, 2021-02-26:
The latest changes introduced syntax that is valid from Python 3.6 but not earlier versions. You seem to be running Python 2.7.
A workaround would be switching to Python 3.6 if you can.

rdica, 2021-02-26:
> The latest changes introduced syntax that is valid from Python 3.6 but not earlier versions. You seem to be running Python 2.7.
> A workaround would be switching to Python 3.6 if you can.

Might want to update README, it says:

# Compatibility with Python versions
This should work on both Python 2 and 3. It was tested with Python 2.7.3 and 3.2.3. The snmp_passpersist mode is Python 2 only because the upstream package is not ready yet.

:smiley:

padelt, 2021-02-26:
You are right, that’s inconsistent and we‘ll try to decide upon a solution in #104 .
Maybe you could comment there if Python 3.6 is an option for you. I‘d like to gather some data what versions people actually use.

ReachableCEO, 2021-02-26:
Oh yeah I can switch. On Ubuntu it’s fine , just Debian buster seems to utilize 2.7 by default (this is all referring to stock system python )

——-
Charles N Wyble
Founder TSYS Group
Email: charles@turnsys.com
Phone: 818 280 7059
Matrix: @Charlesnw:turnsys.com

davet2001, 2021-02-26:
On many distros the default python is 2.7.  But python3 runs a python 3.x version.
Try:
`which python`
`which python3`
To see what happens.

padelt, 2021-03-01:
I guess the situation here is clear, so I'll close.

closed 2021-03-01 by padelt