padelt/temper-python #104: Use of Python 3.6 syntax

padelt, 2021-02-26:
#103 surfaced that we have syntax in the codebase that requires Python 3.6.
At the same time, the Readme mentions compatibility with Python 2.7 and 3.2.

As Python 2 is EOL just as is Python 3.2, I am inclined to up the minimum version to 3.6.

Is there demand for keeping compatibility with earlier versions? Especially 3.5 which seems to be the default in major Linux distributions.

rdica, 2021-02-26:
Works on Fedora 32:
```
ardy@daw3:~$ (0) python -V
Python 3.8.7

ardy@daw3:~$ (0) temper-poll 
Found 1 devices
Device #0: 24.4°C 76.0°F 

```

ps-jay, 2021-02-26:
I would bump the minimum supported version up, for the following reasons:
- it's easier now than in the past to run different versions of Python in parallel (virtualenvs, docker, ...)
- Debian Buster (current latest) has optional Python 3.7 (via the python3 package)
- CentOS 7 has Python 3.6 available via the software collections libraries (and maybe newer, my info is circa 2019 on this)
- This project doesn't usually have the people-power to spend much time on it, and time could probably be spent on "higher value" tasks than supporting EOL's versions of Python

padelt, 2021-03-01:
That's less controversial than I thought.
So let's do it - in #105.

closed 2021-03-01 by padelt