padelt/temper-python #108: ID 0c45:7401 Temperature reading incorrect (TEMPer2V1.4)

BrianGilbert, 2021-05-13:
```
pi@raspberrypi:~/temper-python $ sudo temper-poll -p
WARNING:temperusb.temper:Unrecognised sensor type 'TEMPer2V1.4'. Trying to guess communication format. Please add the configuration to 'device_library.py' and submit to https://github.com/padelt/temper-python/issues to benefit other users.
Found 1 devices
Device #0 (bus 1 - port 3): 28.7°C 83.6°F
```

```
lsusb -v -d 0c45:7401

Bus 001 Device 004: ID 0c45:7401 Microdia TEMPer Temperature Sensor
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0         8
  idVendor           0x0c45 Microdia
  idProduct          0x7401 TEMPer Temperature Sensor
  bcdDevice            0.01
  iManufacturer           1
  iProduct                2
  iSerial                 0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x003b
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.10
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      65
         Report Descriptors:
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval              10
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.10
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      41
         Report Descriptors:
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval              10
pi@raspberrypi:~/temper-python $
```

davet2001, 2021-05-13:
@BrianGilbert Many thanks for this.

I expect the reported reading is valid.  Could be some self heating in the sensor or calibration offset.

I have created a PR #109 for this.  Once merged, the 'unrecognised sensor type' will go away.

padelt, 2021-05-14:
Thanks everyone, I just merged this. Please reopen the issue if needed!

closed 2021-05-14 by padelt