padelt/temper-python #115: support for TEMPer2V1.3 request

coder1995, 2021-11-28:
sudo temper-poll -p
WARNING:temperusb.temper:Unrecognised sensor type 'TEMPer2V1.3'. Trying to guess communication format. Please add the configuration to 'device_library.py' and submit to https://github.com/padelt/temper-python/issues to benefit other users.
Found 1 devices
Device #0 (bus 1 - port 3): 17.9°C 64.3° <--- temperature is off by 40 degrees but this varies from value to value (I can't just simply add 40 to my final result to get reliable readings)

Bus 001 Device 004: ID 0c45:7401 Microdia TEMPer Temperature Sensor
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0         8
  idVendor           0x0c45 Microdia
  idProduct          0x7401 TEMPer Temperature Sensor
  bcdDevice            0.01
  iManufacturer           1
  iProduct                2
  iSerial                 0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           59
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.10
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      65
         Report Descriptors:
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval              10
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.10
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      41
         Report Descriptors:
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval              10

Thank you.

davet2001, 2021-11-29:
@coder1995 Thanks for sharing the info on this new sensor!  Please could you additionally run
```
sudo temper-poll -p -v
```
And paste the output?

Also, please could you answer the following:

1. How many temperature sensors does this device have (usually it is 1)?
2. What was the actual temperature (approx.) of the room you were in?
3. Any other sensors e.g. humidity built in?

If we are really struggling a photo of the outside (and ideally the inside) of the device may help.

coder1995, 2021-11-29:
Here is the output:

sudo temper-poll -p -v
WARNING:temperusb.temper:Unrecognised sensor type 'TEMPer2V1.3'. Trying to guess communication format. Please add the configuration to 'device_library.py' and submit to https://github.com/padelt/temper-python/issues to benefit other users.
DEBUG:temperusb.temper:Found device | Bus:1 Ports:3 SensorCount:2
INFO:temperusb.temper:Found 1 TEMPer devices
Found 1 devices
DEBUG:temperusb.temper:Ctrl transfer: b'\x01\x803\x01\x00\x00\x00\x00'
DEBUG:temperusb.temper:Read data: '80 04 12 d0 12 70 8d 15'
DEBUG:temperusb.temper:Ctrl transfer: b'\x01\x803\x01\x00\x00\x00\x00'
DEBUG:temperusb.temper:Read data: '80 04 12 d0 12 70 8d 15'
DEBUG:temperusb.temper:ID='TEMPer2V1.3'
DEBUG:temperusb.temper:T=18.81250C
DEBUG:temperusb.temper:Ctrl transfer: b'\x01\x803\x01\x00\x00\x00\x00'
DEBUG:temperusb.temper:Read data: '80 04 12 d0 12 70 8d 15'
DEBUG:temperusb.temper:Ctrl transfer: b'\x01\x803\x01\x00\x00\x00\x00'
DEBUG:temperusb.temper:Read data: '80 04 12 d0 12 70 8d 15'
DEBUG:temperusb.temper:ID='TEMPer2V1.3'
Device #0 (bus 1 - port 3): 18.8°C 65.9°F

As for your questions:
1. One sensor, yes.
2. Right now it is installed in a remote location measuring the water temperature from a boiler tank and I don't have direct access to it. However when I did measured against another temperature sensor that worked correctly the reading was: 12.1 (temper) vs 27 (correct).
3. Not as far as I know.

I will take a picture of the outside and the inside as soon as I get to that location but it might take several days.

Thank you very much for your support and work!

davet2001, 2021-11-29:
Ok, thanks for more info.  Sorry for an additional request...

It looks like this device is behaving the same as [other devices](https://github.com/s-leroux/TEMPer2/blob/84e986a89f1d8d4f27473b13176298b0a49632e4/comm.c#L268-L277), i.e. 0x12d0 is converted to floating point and divided by 256 to give value in degC of 18.8125.

Please could you record what you think the temperature is at the time of running `sudo temper-poll -p -v`, then change the temperature (e.g. 20degC higher), record the temperature again, then run the command again?

I'm less interested in what you think the offset might be, and more interested in finding the temperature you are expecting in the returned data.  This will also help me see which bytes are variable.

It is possible that your sensor has a big calibration error (gain or offset or both).

Please note that if it is [this device](https://pcsensor.com/usb-thermometer/temper2.html), there are two sensors, one inside the usb device and one outside on the end of the wire.  Based on the debug data you provided I would say that they are reading 18.8125 and 18.4375C.

coder1995, 2021-11-30:
I will try to do that when I am in that location (it is about 200 km away from me). I will run sudo temper-poll -p -v again when someone will be in that location and will be able to send me a picture of the solar display which shows the correct temperature. Both sensors are inserted in the same cavity in the boiler tank.

It might be very possible that we are getting the "room" temperature from the usb device and not the actual temperature at the end of the wire. That will explain why the value doesn't make any sense as I know how the temperature of the water from the tank varies but it keeps showing normal room temperature levels.

Thank you.

coder1995, 2021-12-02:
This is the temperature reported for 22 C (verified):

pi@raspberrypi:~ $ sudo temper-poll -p -v
WARNING:temperusb.temper:Unrecognised sensor type 'TEMPer2V1.3'. Trying to guess communication format. Please add the configuration to 'device_library.py' and submit to https://github.com/padelt/temper-python/issues to benefit other users.
DEBUG:temperusb.temper:Found device | Bus:1 Ports:3 SensorCount:2
INFO:temperusb.temper:Found 1 TEMPer devices
Found 1 devices
DEBUG:temperusb.temper:Ctrl transfer: b'\x01\x803\x01\x00\x00\x00\x00'
DEBUG:temperusb.temper:Read data: '80 04 0a e0 15 00 1d 15'
DEBUG:temperusb.temper:Ctrl transfer: b'\x01\x803\x01\x00\x00\x00\x00'
DEBUG:temperusb.temper:Read data: '80 04 0a e0 15 00 1d 15'
DEBUG:temperusb.temper:ID='TEMPer2V1.3'
DEBUG:temperusb.temper:T=10.87500C
DEBUG:temperusb.temper:Ctrl transfer: b'\x01\x803\x01\x00\x00\x00\x00'
DEBUG:temperusb.temper:Read data: '80 04 0a e0 15 00 1d 15'
DEBUG:temperusb.temper:Ctrl transfer: b'\x01\x803\x01\x00\x00\x00\x00'
DEBUG:temperusb.temper:Read data: '80 04 0a e0 15 00 1d 15'
DEBUG:temperusb.temper:ID='TEMPer2V1.3'
Device #0 (bus 1 - port 3): 10.9°C 51.6°F

davet2001, 2021-12-02:
The data suggests two temperature readings, 10.875 and 21.0.

So I think it is working.  I have created a pull request #116.

If you know how to clone and run a git branch, then it would be great if you could test it on your hardware.  Let me know if you need any help.

coder1995, 2021-12-02:
Thank you very much for your work. I have pulled the branch and the results are:

sudo temper-poll -p -v
DEBUG:temperusb.temper:Found device | Bus:1 Ports:3 SensorCount:2
INFO:temperusb.temper:Found 1 TEMPer devices
Found 1 devices
DEBUG:temperusb.temper:Ctrl transfer: b'\x01\x803\x01\x00\x00\x00\x00'
DEBUG:temperusb.temper:Read data: '80 04 0b d0 14 b0 1d 15'
DEBUG:temperusb.temper:Ctrl transfer: b'\x01\x803\x01\x00\x00\x00\x00'
DEBUG:temperusb.temper:Read data: '80 04 0b d0 14 b0 1d 15'
DEBUG:temperusb.temper:ID='TEMPer2V1.3'
DEBUG:temperusb.temper:T=11.81250C
DEBUG:temperusb.temper:Ctrl transfer: b'\x01\x803\x01\x00\x00\x00\x00'
DEBUG:temperusb.temper:Read data: '80 04 0b d0 14 b0 1d 15'
DEBUG:temperusb.temper:Ctrl transfer: b'\x01\x803\x01\x00\x00\x00\x00'
DEBUG:temperusb.temper:Read data: '80 04 0b d0 14 b0 1d 15'
DEBUG:temperusb.temper:ID='TEMPer2V1.3'
Device #0 (bus 1 - port 3): 11.8°C 53.3°F

The temperature should be around 22.

Thank you!

davet2001, 2021-12-02:
Ok. There could be a 2nd bug which means only 1 temperature is displaying.
Please could you try this branch which incorporates both the new sensor and an additional fix:
https://github.com/davet2001/temper-python/tree/fix_multisensor_readout

davet2001, 2021-12-02:
Hmm, scratch that.  I think there is something weird going on about how it handles multiple sensors per device.  Let me take a look a bit more and get back to you.

davet2001, 2021-12-02:
Update:
No, go for it. Let me know if it works.  Output should be something like this:
```
Found 1 devices
Reading #0: 0.0°C 32.0°F; 0.0°C 32.0°F 
```

davet2001, 2021-12-02:
@coder1995 you may need to re-fetch that branch as some of it was out of date and i just force-pushed it.
`git fetch` your side should fix it

coder1995, 2021-12-02:
Hmm is I still get only one reading. Is there something that I am missing?

pi@raspberrypi:~/temper-python $ git pull origin fix_multisensor_readout
From https://github.com/davet2001/temper-python
 * branch            fix_multisensor_readout -> FETCH_HEAD
Updating 86af27a..ad523b9
Fast-forward
 .gitignore                  |   1 +
 CHANGELOG.md                |  16 ++++++++
 README.md                   |  79 ++++++++++++++++++++++++++++++++++++--
 temperusb/cli.py            |  10 +++--
 temperusb/device_library.py |  91 ++++++++++++++++++++++++++++++++++++++++++++
 temperusb/snmp.py           |   2 +-
 temperusb/temper.py         |  89 ++++++++++++++++++++++---------------------
 tests/test_temper.py        | 175 ++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
 8 files changed, 412 insertions(+), 51 deletions(-)
 create mode 100644 temperusb/device_library.py
 create mode 100644 tests/test_temper.py
pi@raspberrypi:~/temper-python $ sudo python setup.py install
running install
running bdist_egg
running egg_info
writing temperusb.egg-info/PKG-INFO
writing dependency_links to temperusb.egg-info/dependency_links.txt
writing entry points to temperusb.egg-info/entry_points.txt
writing requirements to temperusb.egg-info/requires.txt
writing top-level names to temperusb.egg-info/top_level.txt
reading manifest file 'temperusb.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
writing manifest file 'temperusb.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-armv7l/egg
running install_lib
running build_py
copying temperusb/snmp.py -> build/lib/temperusb
copying temperusb/temper.py -> build/lib/temperusb
copying temperusb/device_library.py -> build/lib/temperusb
copying temperusb/cli.py -> build/lib/temperusb
creating build/bdist.linux-armv7l/egg
creating build/bdist.linux-armv7l/egg/temperusb
copying build/lib/temperusb/snmp.py -> build/bdist.linux-armv7l/egg/temperusb
copying build/lib/temperusb/temper.py -> build/bdist.linux-armv7l/egg/temperusb
copying build/lib/temperusb/__init__.py -> build/bdist.linux-armv7l/egg/temperusb
copying build/lib/temperusb/device_library.py -> build/bdist.linux-armv7l/egg/temperusb
copying build/lib/temperusb/cli.py -> build/bdist.linux-armv7l/egg/temperusb
byte-compiling build/bdist.linux-armv7l/egg/temperusb/snmp.py to snmp.cpython-37.pyc
byte-compiling build/bdist.linux-armv7l/egg/temperusb/temper.py to temper.cpython-37.pyc
byte-compiling build/bdist.linux-armv7l/egg/temperusb/__init__.py to __init__.cpython-37.pyc
byte-compiling build/bdist.linux-armv7l/egg/temperusb/device_library.py to device_library.cpython-37.pyc
byte-compiling build/bdist.linux-armv7l/egg/temperusb/cli.py to cli.cpython-37.pyc
creating build/bdist.linux-armv7l/egg/EGG-INFO
copying temperusb.egg-info/PKG-INFO -> build/bdist.linux-armv7l/egg/EGG-INFO
copying temperusb.egg-info/SOURCES.txt -> build/bdist.linux-armv7l/egg/EGG-INFO
copying temperusb.egg-info/dependency_links.txt -> build/bdist.linux-armv7l/egg/EGG-INFO
copying temperusb.egg-info/entry_points.txt -> build/bdist.linux-armv7l/egg/EGG-INFO
copying temperusb.egg-info/requires.txt -> build/bdist.linux-armv7l/egg/EGG-INFO
copying temperusb.egg-info/top_level.txt -> build/bdist.linux-armv7l/egg/EGG-INFO
zip_safe flag not set; analyzing archive contents...
creating 'dist/temperusb-1.5.3-py3.7.egg' and adding 'build/bdist.linux-armv7l/egg' to it
removing 'build/bdist.linux-armv7l/egg' (and everything under it)
Processing temperusb-1.5.3-py3.7.egg
Removing /usr/local/lib/python3.7/site-packages/temperusb-1.5.3-py3.7.egg
Copying temperusb-1.5.3-py3.7.egg to /usr/local/lib/python3.7/site-packages
temperusb 1.5.3 is already the active version in easy-install.pth
Installing temper-poll script to /usr/local/bin
Installing temper-snmp script to /usr/local/bin

Installed /usr/local/lib/python3.7/site-packages/temperusb-1.5.3-py3.7.egg
Processing dependencies for temperusb==1.5.3
Searching for pyusb==1.2.1
Best match: pyusb 1.2.1
Processing pyusb-1.2.1-py3.7.egg
pyusb 1.2.1 is already the active version in easy-install.pth

Using /usr/local/lib/python3.7/site-packages/pyusb-1.2.1-py3.7.egg
Finished processing dependencies for temperusb==1.5.3
pi@raspberrypi:~/temper-python $ sudo temper-poll
Found 1 devices
Device #0: 11.6°C 52.9°F

coder1995, 2021-12-02:
pi@raspberrypi:~/temper-python $ git pull origin 7d101acc6eace3cafb4f15b9f81b4c28bb7f3d42
From https://github.com/davet2001/temper-python
 * branch            7d101acc6eace3cafb4f15b9f81b4c28bb7f3d42 -> FETCH_HEAD
Merge made by the 'recursive' strategy.
pi@raspberrypi:~/temper-python $ sudo python setup.py install

after setup.py:

pi@raspberrypi:~/temper-python $ sudo temper-poll
Found 1 devices
WARNING:temperusb.temper:Encountered [Errno 19] No such device (it may have been disconnected), resetting <DEVICE ID 0c45:7401 on Bus 001 Address 004> and trying again.
Device #0: 11.6°C 52.9°F

davet2001, 2021-12-02:
Ah, I think  see what is wrong.  Default behavior of command line is to only read sensor 0.
Try running
`sudo temper-poll -p -v -s all`

coder1995, 2021-12-02:
sudo temper-poll -p -v -s all
DEBUG:temperusb.temper:Found device | Bus:1 Ports:3 SensorCount:2
INFO:temperusb.temper:Found 1 TEMPer devices
Found 1 devices
DEBUG:temperusb.temper:Ctrl transfer: b'\x01\x803\x01\x00\x00\x00\x00'
DEBUG:temperusb.temper:Read data: '80 04 0b 60 14 50 1d 15'
DEBUG:temperusb.temper:Ctrl transfer: b'\x01\x803\x01\x00\x00\x00\x00'
DEBUG:temperusb.temper:Read data: '80 04 0b 60 14 50 1d 15'
DEBUG:temperusb.temper:ID='TEMPer2V1.3'
DEBUG:temperusb.temper:T=11.37500C
DEBUG:temperusb.temper:T=20.31250C
DEBUG:temperusb.temper:Ctrl transfer: b'\x01\x803\x01\x00\x00\x00\x00'
DEBUG:temperusb.temper:Read data: '80 04 0b 60 14 50 1d 15'
DEBUG:temperusb.temper:Ctrl transfer: b'\x01\x803\x01\x00\x00\x00\x00'
DEBUG:temperusb.temper:Read data: '80 04 0b 60 14 50 1d 15'
DEBUG:temperusb.temper:ID='TEMPer2V1.3'
Device #0 (bus 1 - port 3): 11.4°C 52.5°F; 20.3°C 68.6°F

Brilliant! It works! Thank you very much for your help!

davet2001, 2021-12-02:
Ok, great!  When the PR gets merged, and then later released as a new version of temper-python it will be available to everyone.  Until then it is ok to keep using your local git copy.  

Also you can use `sudo temper-poll -s 1` to just read the 2nd sensor if you want.

coder1995, 2021-12-02:
Actually I can surely use the first sensor because it reads the general temperature in the boiler room - which is good to have. It is so cool that now I can monitor the water temperature from the boiler remotely. Thanks!

[image]

davet2001, 2021-12-04:
This issue was fixed by #116

closed 2021-12-04 by davet2001