padelt/temper-python #120: Sorry, not an issue - adding V1.1 support

bjornenki, 2022-04-03:
Hi, thanks for this project! 

Sorry it's not a pull request, but in case it helps others, I just wanted to say that I have some old V1.1 units that work just fine with the following in device_library.py:

    "TEMPerHumiV1.1": TemperConfig(
        temp_sens_offsets=[2],
        hum_sens_offsets=[4],
        type=TemperType.FM75,
    ),

Sample output:

Found 2 devices
Device #0: 25.7°C 78.2°F 46.2%RH
Device #1: 25.6°C 78.0°F 46.1%RH

davet2001, 2022-04-03:
Many thanks for your contribution!
If no one else gets to it first, I will create a PR in a few days time 👍

davet2001, 2022-04-18:
> Sample output:
> 
> Found 2 devices 
> Device #0: 25.7°C 78.2°F 46.2%RH
> Device #1: 25.6°C 78.0°F 46.1%RH

@bjornenki Did you have 2 devices plugged in at the time, or does the one USB device have two T and two RH sensors?

bjornenki, 2022-04-18:
Yes, sorry for the confusion, but I did have two different 1.0 devices
plugged in at once.

closed 2022-04-18 by davet2001