padelt/temper-python #128: Support for TEMPerGold (1a86:e025)

allo-, 2022-11-27:
I bought a new TEMPer and tried to get v1 models because I was unsure about the support for v2, but it still seems to be a newer model:

    1a86:e025 QinHeng Electronics TEMPerGold

Any chance to get it to work?

davet2001, 2022-11-27:
@allo- If you are ok with doing this, you could clone the repository and add
`(0x1a86, 0xe025)`
to `VIDPIDS` in `temper.py` then see what happens when you run `python3 temperusb/cli.py -v`.

It should detect your device then try to guess the communication format.

If you get a sensible looking value for temperature, it will be quite straightforward to add support.

allo-, 2022-11-27:
Currently I get an import error when trying to run `cli.py`

```
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/temperusb/cli.py", line 6, in <module>
    from .temper import TemperHandler
ImportError: attempted relative import with no known parent package
```

davet2001, 2022-11-27:
Try `python3 -m temperusb.cli -v`

allo-, 2022-11-27:
It doesn't seem to work out of the box

```
WARNING:temperusb.temper:Unrecognised sensor type 'TEMPerGold'. Trying to guess communication format. Please add the configuration to 'device_library.py' and submit to https://github.com/padelt/temper-python/issues to benefit other users.
DEBUG:temperusb.temper:Found device | Bus:1 Ports:2 SensorCount:2
INFO:temperusb.temper:Found 1 TEMPer devices
Found 1 devices
DEBUG:temperusb.temper:Ctrl transfer: b'\x01\x803\x01\x00\x00\x00\x00'
WARNING:temperusb.temper:Encountered [Errno 110] Operation timed out, resetting <DEVICE ID 1a86:e025 on Bus 001 Address 008> and trying again.
DEBUG:temperusb.temper:Ctrl transfer: b'\x01\x803\x01\x00\x00\x00\x00'
ERROR:temperusb.temper:[Errno 110] Operation timed out
Traceback (most recent call last):
  File "/root/temper-python/temperusb/temper.py", line 249, in get_data
    self._interrupt_read()
  File "/root/temper-python/temperusb/temper.py", line 415, in _interrupt_read
    data = self._device.read(ENDPOINT, REQ_INT_LEN, timeout=TIMEOUT)
  File "/usr/local/lib/python3.9/dist-packages/usb/core.py", line 1029, in read
    ret = fn(
  File "/usr/local/lib/python3.9/dist-packages/usb/backend/libusb1.py", line 864, in intr_read
    return self.__read(self.lib.libusb_interrupt_transfer,
  File "/usr/local/lib/python3.9/dist-packages/usb/backend/libusb1.py", line 954, in __read
    _check(retval)
  File "/usr/local/lib/python3.9/dist-packages/usb/backend/libusb1.py", line 602, in _check
    raise USBTimeoutError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBTimeoutError: [Errno 110] Operation timed out

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.9/runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.9/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/root/temper-python/temperusb/cli.py", line 110, in <module>
    main()
  File "/root/temper-python/temperusb/cli.py", line 59, in main
    temperatures = dev.get_temperatures(sensors=sensors)
  File "/root/temper-python/temperusb/temper.py", line 329, in get_temperatures
    data = self.get_data()
  File "/root/temper-python/temperusb/temper.py", line 279, in get_data
    return self.get_data(True)
  File "/root/temper-python/temperusb/temper.py", line 249, in get_data
    self._interrupt_read()
  File "/root/temper-python/temperusb/temper.py", line 415, in _interrupt_read
    data = self._device.read(ENDPOINT, REQ_INT_LEN, timeout=TIMEOUT)
  File "/usr/local/lib/python3.9/dist-packages/usb/core.py", line 1029, in read
    ret = fn(
  File "/usr/local/lib/python3.9/dist-packages/usb/backend/libusb1.py", line 864, in intr_read
    return self.__read(self.lib.libusb_interrupt_transfer,
  File "/usr/local/lib/python3.9/dist-packages/usb/backend/libusb1.py", line 954, in __read
    _check(retval)
  File "/usr/local/lib/python3.9/dist-packages/usb/backend/libusb1.py", line 602, in _check
    raise USBTimeoutError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBTimeoutError: [Errno 110] Operation timed out

```

davet2001, 2022-11-27:
Ok. Looks like it needs a different command message to make it respond.

Could be tricky to get this working unless you are able to get deep into debugging it, possibly by looking at other libraries or sniffing the traffic when it's communicating with an app that works.

allo-, 2022-11-27:
I am not experienced in it and don't have the time to do things like usb logging on windows. But when you have concrete things I can just run on my raspberry pi I'd happily try them.

davet2001, 2022-11-28:
Probably this needs a developer with access to the hardware to make it work.

If you can find an example of this working in another library, please let me know.

davet2001, 2023-06-09:
For info: In #130 we added support for a USB device with the same VID & PID as this PR asks for, so should be a step closer.

Stylopidae1793, 2023-07-18:
I am having a similar issue, but in this case the program does not detect a TEMPER2 device with the model number 3553:a001. It detects all previous model numbers, however.

I added the model numbers to the firmware identifier portion of the code, and it still does not detect. However, it still detects my other TEMPER units which have a different model number.

They recently updated the model numbers for their units, so I suspect others may be having this problem as well.

davet2001, 2023-07-18:
Hi @Stylopidae1793, `3553:a001` looks more like a vendor id and pid but I can't seem to find it looking on a few databases.

The library code identifies the type by looking at the USB identifier and the product name reported by the device such as 'TEMPerGold'.

If any of these don't exactly match then the device is ignored or guessed.

You can get more info about the device using 

`lsusb`
and
`dmesg | grep input`

You may need to [set the device permissions](https://github.com/padelt/temper-python/blob/master/README.md#usb-device-permissions) and reboot.

Stylopidae1793, 2023-07-19:
Thanks for responding.

Just to give a little background, I am using multiple TEMPER2 units to monitor plant growth conditions inside a growth chamber. Like allo-, I'm very new to programming and get lost pretty easily.

I got the information above by using `lsusb`. For the working units, they give me the vendor/PID as '1a86:e025'. For the nonworking units, they give me the number '3553:a001'. When the system is pinged, there is no error and this tracks with what you said above.

When I use lsusb, this is what I get on the nonworking unit:

'ID 3553:a001 PCsensor TEMPer2'

On the working unit, I get:

'ID 1a86:e025 QinHeng Electronics TEMPer2'

What am I looking for when I use 'dmesg | grep input'? It gives me quite a bit of information, and includes the vendor ID for both working and nonworking units. I'd be happy to post a screenshot, if needed.

As for setting the device permissions, this specific program does not contain a file titled '99-tempsensor.rules', so when I try to install 'sudo cp etc/99-tempsensor.rules /etc/udev/rules.d/', I get a 'cannot stat' error. Googling the error, it tells me that it's having trouble copying a folder that's not included in the program, although it's possible I've misunderstood something along the way.

davet2001, 2023-07-19:
@Stylopidae1793 not sure if you replied but then deleted your message.
Anyway, you need to get the 'product name'.  This is not the shortened name such as Temper or Temper2 listed by lsusb, but the full name with version number etc.
example output
```
$ sudo dmesg | grep input

[168543.553353] input: RDing TEMPERHUM1V1.3 as /devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.0/0003:0C45:7402.0004/input/input26
[168543.610756] hid-generic 0003:0C45:7402.0004: input,hidraw3: USB HID v1.10 Keyboard [RDing TEMPERHUM1V1.3] on usb-0000:00:14.0-4/input0
[168543.614072] hid-generic 0003:0C45:7402.0005: hiddev0,hidraw4: USB HID v1.10 Device [RDing TEMPERHUM1V1.3] on usb-0000:00:14.0-4/input1
```
In my case, that means the product name is TEMPERHUM1V1.3

Please create a new issue for this though.  In the issue description, please paste the output of 
`sudo lsusb -d 3553:a001 -v`

Stylopidae1793, 2023-07-20:
Yes! I had replied, but realized I didn't check my code thoroughly enough and ended up finding a few typos. So I deleted my response and went to double/triple check.

The units are working now; the key was to update the device permissions and to update the firmware section of the code.

davet2001, 2023-07-20:
Ok, please can you summarise the changes you made to get it working in order that a PR can be created so that others can benefit?