padelt/temper-python #129: TEMPer2 not detecting devices

Stylopidae1793, 2023-04-27:
Hi,

I'm trying to get the Temper2 device to work with this code.

Basically, when I run 'temper-poll', I get an error saying that it found 0 devices. However, the 'txt' button works.

The device model is 1a86:e025.

I'm new to raspberry pis in general, and python specifically, and I didn't see anything which resembled this problem on other github pages.

davet2001, 2023-05-02:
Ok, the device you list is made by [QinHeng Electronics](https://usb-ids.gowdy.us/read/UD/1a86) and since it is not listed on at least two PID databases, I'm guessing it's either new or obscure.

Doing a bit of digging, and looking at [another implementation](https://github.com/ccwienk/temper/commit/0178978cccd8e5ee14d67207e91c8cf78196e1d3), it's possible this is a TEMPer2_V3.7 with two temperature sensors, one internal and one external.  Does that sound about right?

If I create a pull request for this, do you have the capability to clone it and try it?

davet2001, 2023-05-02:
@Stylopidae1793 or any other volunteer who has this device: please could you try the above PR to see if it works with your sensor?

Stylopidae1793, 2023-05-02:
Yes! I'd be happy to clone the pull request and try it!

Stylopidae1793, 2023-05-03:
Using the lsusb function, this is indeed produced by QinHeng Electronics and has the external and internal sensor that you described. However, I can't verify the model (or version) because it doesn't appear on the packaging or in the programming as far as I can tell.

I cloned the repository above and followed the instructions in the README and unfortunately it still says 'Found 0 devices' when I use the temper-poll command.

davet2001, 2023-05-04:
@Stylopidae1793 thanks for this. Please can you paste the output of `lsusb -v -d 1a86:e025`?

Then just to confirm that you do have the correct branch, could you also paste the output of `git status` and `python3 -m temperusb.cli -v`?  Many thanks!

Stylopidae1793, 2023-05-05:
Here's the output you requested:

```
lsusb -v -d 1a86:e025

Bus 001 Device 007: ID 1a86:e025 QinHeng Electronics TEMPer2
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x1a86 QinHeng Electronics
  idProduct          0xe025 
  bcdDevice            1.00
  iManufacturer           1 PCsensor
  iProduct                2 TEMPer2
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0042
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      62
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.10
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      23
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval              10
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
```
```
git status
On branch temper2_new
Your branch is up to date with 'origin/temper2_new'.

nothing to commit, working tree clean
```
```
python3 -m temperusb.cli -v
Traceback (most recent call last):
  File "/home/pi/temper-python/temperusb/temper.py", line 99, in __init__
    productname = self._device.product
  File "/usr/local/lib/python3.9/dist-packages/pyusb-1.2.1-py3.9.egg/usb/core.py", line 875, in product
    self._product = util.get_string(self, self.iProduct)
  File "/usr/local/lib/python3.9/dist-packages/pyusb-1.2.1-py3.9.egg/usb/util.py", line 313, in get_string
    raise ValueError("The device has no langid"
ValueError: The device has no langid (permission issue, no string descriptors supported or device error)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.9/runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.9/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/home/pi/temper-python/temperusb/cli.py", line 110, in <module>
    main()
  File "/home/pi/temper-python/temperusb/cli.py", line 42, in main
    th = TemperHandler()
  File "/home/pi/temper-python/temperusb/temper.py", line 432, in __init__
    self._devices += [TemperDevice(device) for device in \
  File "/home/pi/temper-python/temperusb/temper.py", line 432, in <listcomp>
    self._devices += [TemperDevice(device) for device in \
  File "/home/pi/temper-python/temperusb/temper.py", line 102, in __init__
    raise usb.core.USBError("Error reading langids from device. "+
usb.core.USBError: [Errno None] Error reading langids from device. This might be a permission issue. Please check that the device node for your TEMPer devices can be read and written by the user running this code. The temperusb README.md contains hints about how to fix this. Search for 'USB device permissions'.
```

Stylopidae1793, 2023-05-08:
It doesn't look like the code I added is showing up.

```
cat /etc/udev/rules.d/99-tempsensor.rules
SUBSYSTEMS=="usb", ACTION=="add", ATTRS{idVendor}=="0c45", ATTRS{idProduct}=="7401", MODE="666"
SUBSYSTEMS=="usb", ACTION=="add", ATTRS{idVendor}=="0c45", ATTRS{idProduct}=="7402", MODE="666"
```

```python3 -m temperusb.cli -v
Traceback (most recent call last):
  File "/home/pi/temper-python/temperusb/temper.py", line 99, in __init__
    productname = self._device.product
  File "/usr/local/lib/python3.9/dist-packages/pyusb-1.2.1-py3.9.egg/usb/core.py", line 875, in product
    self._product = util.get_string(self, self.iProduct)
  File "/usr/local/lib/python3.9/dist-packages/pyusb-1.2.1-py3.9.egg/usb/util.py", line 313, in get_string
    raise ValueError("The device has no langid"
ValueError: The device has no langid (permission issue, no string descriptors supported or device error)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.9/runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.9/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/home/pi/temper-python/temperusb/cli.py", line 110, in <module>
    main()
  File "/home/pi/temper-python/temperusb/cli.py", line 42, in main
    th = TemperHandler()
  File "/home/pi/temper-python/temperusb/temper.py", line 432, in __init__
    self._devices += [TemperDevice(device) for device in \
  File "/home/pi/temper-python/temperusb/temper.py", line 432, in <listcomp>
    self._devices += [TemperDevice(device) for device in \
  File "/home/pi/temper-python/temperusb/temper.py", line 102, in __init__
    raise usb.core.USBError("Error reading langids from device. "+
usb.core.USBError: [Errno None] Error reading langids from device. This might be a permission issue. Please check that the device node for your TEMPer devices can be read and written by the user running this code. The temperusb README.md contains hints about how to fix this. Search for 'USB device permissions'.```

davet2001, 2023-05-08:
You definitely need to edit the udev rules file until you see 3 lines in it.

What happened when you tried to save it?

davet2001, 2023-05-08:
Try
`sudo echo '"SUBSYSTEMS=="usb", ACTION=="add", ATTRS{idVendor}=="1a86", ATTRS{idProduct}=="e025", MODE="666"' >> /etc/udev/rules.d/99-tempsensor.rules`

Then check it using the same method.

Stylopidae1793, 2023-05-09:
It still says permission denied, unfortunately.

davet2001, 2023-05-09:
You need to be able to modify the config.

Can you try the following in a terminal:

`sudo nano /etc/udev/rules.d/99-tempsensor.rules`

Then add the new line followed by enter

Then press ctrl-o to write the file?

Stylopidae1793, 2023-05-10:
That worked! The new line of code shows up in the 99-tempsensor.rules file. It doesn't let me copy/paste it in it's entirety, but it's definitely there.

temper-poll still gives me an error, though:

```
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/temperusb-1.6.0-py3.9.egg/temperusb/temper.py", line 99, in __init__
  File "/usr/local/lib/python3.9/dist-packages/pyusb-1.2.1-py3.9.egg/usb/core.py", line 875, in product
    self._product = util.get_string(self, self.iProduct)
  File "/usr/local/lib/python3.9/dist-packages/pyusb-1.2.1-py3.9.egg/usb/util.py", line 313, in get_string
    raise ValueError("The device has no langid"
ValueError: The device has no langid (permission issue, no string descriptors supported or device error)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/bin/temper-poll", line 33, in <module>
    sys.exit(load_entry_point('temperusb==1.6.0', 'console_scripts', 'temper-poll')())
  File "/usr/local/lib/python3.9/dist-packages/temperusb-1.6.0-py3.9.egg/temperusb/cli.py", line 42, in main
  File "/usr/local/lib/python3.9/dist-packages/temperusb-1.6.0-py3.9.egg/temperusb/temper.py", line 432, in __init__
  File "/usr/local/lib/python3.9/dist-packages/temperusb-1.6.0-py3.9.egg/temperusb/temper.py", line 432, in <listcomp>
  File "/usr/local/lib/python3.9/dist-packages/temperusb-1.6.0-py3.9.egg/temperusb/temper.py", line 102, in __init__
usb.core.USBError: [Errno None] Error reading langids from device. This might be a permission issue. Please check that the device node for your TEMPer devices can be read and written by the user running this code. The temperusb README.md contains hints about how to fix this. Search for 'USB device permissions'.
```

davet2001, 2023-05-10:
@Stylopidae1793 Did you reboot after editing the file? If not please can you reboot and retry?

davet2001, 2023-05-10:
Please can you also paste the output from `cat /etc/udev/rules.d/99-tempsensor.rules`?

Stylopidae1793, 2023-05-11:
I edited the file, and then rebooted the device. I shut the pi down, then restarted it. I also used ```sudo reboot```.

Here's the output:
```
cat /etc/udev/rules.d/99-tempsensor.rules
SUBSYSTEMS=="usb", ACTION=="add", ATTRS{idVendor}=="0c45", ATTRS{idProduct}=="7401", MODE="666"
SUBSYSTEMS=="usb", ACTION=="add", ATTRS{idVendor}=="0c45", ATTRS{idProduct}=="7402", MODE="666"
SUBSYSTEMS=="usb", ACTION=="add", ATTRS{idVendor}=="1a86", ATTRS{idProduct}=="e025", MODE="666"
```

davet2001, 2023-05-11:
Ok, please can you confirm that you still got the error after the reboot?

If so, please can you try running the script as root, I `sudo python3 -m temperusb.cli -v`?

You may need to install some extra packages via sudo pip.  If that happens, I would appreciate if you could resolve those issues yourself via Google before coming back with a question.

(If you can confirm the library works, I can merge it and resolve your permissions issue outside of this ticket.)

davet2001, 2023-06-04:
@Stylopidae1793 did you get any success testing this?

Stylopidae1793, 2023-06-06:
My apologies for the long delay.

I did some troubleshooting on my own, and I was finally able to get it working last week. I'm not entirely sure what I did, but eventually it started working.

I'll close the thread now.

closed 2023-06-06 by Stylopidae1793

davet2001, 2023-06-06:
@Stylopidae1793 if the new branch worked, I'll merge it so that everyone can use it.

Do I understand correctly that the previously non working TEMPer device now works with the code from the branch that I linked above?