padelt/temper-python #134: Support for TEMPerGold (3553:a001)

BeyondAeon, 2023-10-30:
Hi I've Picked up 3 Temper sensors & Tried to get them working.
I've  added the USB ID to the devices
It Detects the Device and then Raises A Timeout and Dies.
I'm not great with Python or USB Protocol, But I have a Wireshark USB cap of the windows Software talking to the device. 

I can provide the capture if it would help....

davet2001, 2023-10-30:
Thanks @BeyondAeon!

Please can you 
1. summarise the code you added when you tried
2. Run the cli with -v and copy & paste the output
3. Attach the capture file when you have it working in windows
4. Tell us the real world reading that windows reported so we can spot them in the USB traffic

Thanks for contributing!

BeyondAeon, 2023-10-30:
Hi, The cli.py doesn't work for me, gives me a error, 
"import TemperHandler
ImportError: attempted relative import with no known parent package"
(I'm a python newb still)

added
    "TEMPerGold": TemperConfig(
        temp_sens_offsets=[2, 4],
        hum_sens_offsets=None,
        type=TemperType.FM75,
    ),
to device_library.py

added     (0x3553, 0xa001),   to VIDPIDS in temper.py

Attached zip file with Windows wireshark USB capture and screenshot of windows program.

[tempercap.zip](https://github.com/padelt/temper-python/files/13210244/tempercap.zip)

monkeypashion, 2023-12-09:
I have a new TEMPer2 (also 3553:a001) and I get the exact same issue. 

(myenv) pi@raspberrypi-stove-v2:~/temper-python $ **lsusb**
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
**Bus 001 Device 004: ID 3553:a001 PCsensor TEMPer2**
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

I have also made the changes:

added (0x3553, 0xa001) to VIDPIDS in temper.py
added the following to DEVICE_LIBRARY in device_library.py
    "TEMPer2": TemperConfig(
        temp_sens_offsets=[2, 4],
        hum_sens_offsets=None,
        type=TemperType.FM75,
    ),

After these changes it finds the device but it fails to read the data. Here the output from temperusb.cli:

(myenv) pi@raspberrypi-stove-v2:~/temper-python $ **sudo /home/pi/myenv/bin/python -m temperusb.cli -v**
DEBUG:temperusb.temper:Found device | Bus:1 Ports:3 SensorCount:2
INFO:temperusb.temper:Found 1 TEMPer devices
Found 1 devices
DEBUG:temperusb.temper:Ctrl transfer: b'\x01\x803\x01\x00\x00\x00\x00'
WARNING:temperusb.temper:Encountered [Errno 110] Operation timed out, resetting <DEVICE ID 3553:a001 on Bus 001 Address 004> and trying again.
DEBUG:temperusb.temper:Ctrl transfer: b'\x01\x803\x01\x00\x00\x00\x00'
ERROR:temperusb.temper:[Errno 110] Operation timed out
Traceback (most recent call last):
  File "/home/pi/temper-python/temperusb/temper.py", line 256, in get_data
    self._interrupt_read()
  File "/home/pi/temper-python/temperusb/temper.py", line 422, in _interrupt_read
    data = self._device.read(ENDPOINT, REQ_INT_LEN, timeout=TIMEOUT)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pi/myenv/lib/python3.11/site-packages/usb/core.py", line 1029, in read
    ret = fn(
          ^^^
  File "/home/pi/myenv/lib/python3.11/site-packages/usb/backend/libusb1.py", line 864, in intr_read
    return self.__read(self.lib.libusb_interrupt_transfer,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pi/myenv/lib/python3.11/site-packages/usb/backend/libusb1.py", line 954, in __read
    _check(retval)
  File "/home/pi/myenv/lib/python3.11/site-packages/usb/backend/libusb1.py", line 602, in _check
    raise USBTimeoutError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBTimeoutError: [Errno 110] Operation timed out

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/home/pi/temper-python/temperusb/cli.py", line 110, in <module>
    main()
  File "/home/pi/temper-python/temperusb/cli.py", line 59, in main
    temperatures = dev.get_temperatures(sensors=sensors)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pi/temper-python/temperusb/temper.py", line 336, in get_temperatures
    data = self.get_data()
           ^^^^^^^^^^^^^^^
  File "/home/pi/temper-python/temperusb/temper.py", line 286, in get_data
    return self.get_data(True)
           ^^^^^^^^^^^^^^^^^^^
  File "/home/pi/temper-python/temperusb/temper.py", line 256, in get_data
    self._interrupt_read()
  File "/home/pi/temper-python/temperusb/temper.py", line 422, in _interrupt_read
    data = self._device.read(ENDPOINT, REQ_INT_LEN, timeout=TIMEOUT)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pi/myenv/lib/python3.11/site-packages/usb/core.py", line 1029, in read
    ret = fn(
          ^^^
  File "/home/pi/myenv/lib/python3.11/site-packages/usb/backend/libusb1.py", line 864, in intr_read
    return self.__read(self.lib.libusb_interrupt_transfer,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pi/myenv/lib/python3.11/site-packages/usb/backend/libusb1.py", line 954, in __read
    _check(retval)
  File "/home/pi/myenv/lib/python3.11/site-packages/usb/backend/libusb1.py", line 602, in _check
    raise USBTimeoutError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBTimeoutError: [Errno 110] Operation timed out

Any help much appreciated :)

by, 2024-07-06:
I do have the very same problem as https://github.com/padelt/temper-python/issues/134#issuecomment-1848606785 with my new TEMPer2V4.1 <DEVICE ID 3553:a001>, this is what I treid: https://github.com/padelt/temper-python/compare/master...by:temper-python:master
Thank you very much for your help!

by, 2024-07-06:
P.S.: Here is the reply of `$ sudo temper --verbose`
```
Firmware query: b'0186ff0100000000'
Firmware value: b'54454d506572325f56342e3100000000' TEMPer2_V4.1
Data value: b'80800b544e200000800109ca4e200000'
Converted value: b'09ca'
Bus 003 Dev 002 3553:a001 TEMPer2_V4.1 29.00C 84.20F - 25.06C 77.11F -
```

by, 2024-07-14:
P.S.2: Solved by returning the device.