padelt/temper-python #135: Can you do a release with all the fixes from the last 2 years?

deftdawg, 2023-12-18:
Hi, could you kindly push out a new release containing the enhancements and fixes from the past 2 years?   

In Home Assistant, attempting to use `TEMPerHumiV1.0` I get the following error:
```
Logger: temperusb.temper
Source: components/temper/sensor.py:41
First occurred: 18:56:52 (1 occurrences)
Last logged: 18:56:52

Unrecognised sensor type 'TEMPerHumiV1.0'. Trying to guess communication format. Please add the configuration to 'device_library.py' and submit to https://github.com/padelt/temper-python/issues to benefit other users.
```

Support for `TEMPerHumiV1.0` was added more than a year ago with the merge of #124 and I've verified that it works fine in master on my Raspberry Pi with the following:
```sh
cd ~/config
git clone https://github.com/padelt/temper-python
cd temper-python
python -m venv venv
. venv/bin/activate
python setup.py install
sudo venv/bin/temper-poll
# Found 1 devices
# Device #0: 28.4C 83.1F 30.4%RH
```

It just needs to be made available as an update `temperusb` pypi package for Home Assistant to consume:

https://github.com/home-assistant/core/blob/017dc6604656331022069bdab7471c371fecc0e2/homeassistant/components/temper/manifest.json#L8C21-L8C30

Thanks and :beers:

davet2001, 2023-12-18:
@deftdawg yes, i agree, a release would be a good idea. Unfortunately I don't have access to do it. 
@padelt can you help?

padelt, 2023-12-18:
Could I add you as a PyPI owner of the project @davet2001 ? Do you have a handle/username there?

deftdawg, 2023-12-18:
@amorphic is also listed as a PyPi maintainer, he might be able to help get a new temperusb release out if @davet2001 isn't yet on PyPi.

davet2001, 2023-12-18:
I am on pypi, same username.

padelt, 2023-12-18:
PyPI tells me I can't add you @davet2001 because the user is missing a valid email address. Could you add one and let me know?

davet2001, 2023-12-18:
> Could you add one and let me know?

Done. I had not used pypi before to publish packages.

deftdawg, 2023-12-21:
Any luck with this @davet2001, anything I can do to help?

davet2001, 2023-12-21:
Yes, in progress. I have built the release but had to spend last night unblocking drains!
Need to upload to testpypi then pypi. Probably will be tonight.

davet2001, 2023-12-21:
@padelt Please can you also add me to the collaborators for test.pypi.org? I.e.
https://test.pypi.org/manage/project/temperusb/collaboration/
I want to check it works before i put it on the real pypi!

padelt, 2023-12-21:
Done!

davet2001, 2023-12-21:
1.6.1 Now published to pypi
https://pypi.org/project/temperusb/1.6.1/

closed 2023-12-21 by davet2001

deftdawg, 2023-12-22:
Thanks so much @davet2001 and @padelt

padelt, 2023-12-22:
Thank you @davet2001 !!