padelt/temper-python #139: New sensor type

safrye, 2024-08-02:
Hi,
I got the warning about a new sensor type:
`WARNING:temperusb.temper:Unrecognised sensor type 'TEMPer2_M12_V1.3'. Trying to guess communication format. Please add the configuration to 'device_library.py' and submit to https://github.com/padelt/temper-python/issues to benefit other users.`
Lsusb showed:
`Bus 003 Device 002: ID 0c45:7401 Microdia TEMPer Temperature Sensor`
This is a metal body TEMPer2 with a TXT button.

Cheers.

davet2001, 2024-08-02:
Thanks @safrye! Did it connect ok and give a valid temperature reading?

safrye, 2024-08-03:
It connected fine after I installed the USB permission and gave valid reading.
Interestingly, also the old pcsensor program did not work before I installed the permissions, which was also the reason why I looked for a newer version. I use OpenSuse Tumbleweed and at least a year ago pcsensor didn't complain about not finding the device.
Just to be complete here also the output of the TXT feature, without the external probe:
<pre>
www.pcsensor.com
  type:fm75
   caps lock:on/off/++ fw:1.3
    num lock:Off/on/--  
    date	time	c	c	interval
			inside	outside
					25.50	err	1
							25.50	err	1
									25.63	err	1
</pre>
Cheers and
Thanks for the nice program.

P.S.: I recognized that temper-python only gives one decimal place for the temperature while pcsensor gives two. Any special reason for this?

davet2001, 2024-08-18:
@safrye Regarding the 1decimal place output, this is just how the output is formatted in the statement that prints the number to the screen, e.g. https://github.com/padelt/temper-python/blob/67d544dcf1bfa46726f88fe3cba2936cf7842029/temperusb/cli.py#L82

If you want more decimal places, you could import the library, create a `TemperDevice()` instance, then call `get_temperature()`. This will return a floating point value.

But do bear in mind the finite resolution, noise, measurement uncertainty and calibration error of these sensors if looking at the second decimal point.

davet2001, 2025-01-09:
Support for this sensor will be available in 1.6.2 when it is released.

PR #142 (2025-01-09): Add TEMPer2_M12_V1.3
Add new sensor type with tests to fix #139
closed 2025-01-09 by davet2001