padelt/temper-python #141: Python 3 doesn't support unbuffered text I/O

EdwardBetts, 2025-01-04:
`temper-snmp` tries to do unbuffered text I/O, but this is not supported in Python 3. Presumably nobody is using the SNMP feature and this broken code should be removed.

```
$ python3.13 temperusb/snmp.py 
Traceback (most recent call last):
  File "/home/edward/src/2025/vendor/temper-python/temperusb/snmp.py", line 94, in <module>
    main()
    ~~~~^^
  File "/home/edward/src/2025/vendor/temper-python/temperusb/snmp.py", line 86, in main
    sys.stdout = _unbuffered_handle(sys.stdout)
                 ~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^
  File "/home/edward/src/2025/vendor/temper-python/temperusb/snmp.py", line 21, in _unbuffered_handle
    return os.fdopen(fd.fileno(), 'w', 0)
           ~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^
  File "<frozen os>", line 1068, in fdopen
ValueError: can't have unbuffered text I/O
$ ```

davet2001, 2025-01-04:
I tend to agree but @padelt should comment on this.

padelt, 2025-01-04:
I agree, too. Looks like it is not used with recent Python versions at all.
This means we should direct the effort to removing SNMP functionality instead of trying to fix it.